Noam-Dori/ros-integrate #5: Wrongly take my junit run button.

hliao, 2019-04-30:
Hi 

I am using intellij for java, and all the junit test case button had been replace by your plugin , when I click on it ,it create a test run with the cute icon , which actually I want to run my test case via junit. please help fix it, it is really boring.  

BR
Xavier

hliao, 2019-04-30:
And something really interesting , I didn't install this plugin , . maybe it is not a bug from your side, please help check.

Noam-Dori, 2019-05-14:
What icon is shown? Could you post a picture of it? I am afraid I don't understand what icon you refer to when you say "cute icon". Moreover, if you didn't install this plugin, its highly unlikely that this plugin is the cause of it.

closed 2019-10-03 by Noam-Dori