Noam-Dori/ros-integrate #10: The specified source space does not exist

nbro, 2019-10-15:
I am getting the following error when I attempt to run a Python module

```
sh -c "echo 'ROS workspace directory: /home/nbro' &&
      cd /home/nbro &&
      catkin_make &&
      echo 'Sourcing /home/nbro:devel:setup.sh' &&
      . /home/nbro:devel:setup.sh &&
      echo 'Available nodes:' &&
      rosnode list &&
      echo 'Available topics:' &&
      rostopic list &&
      echo 'Available services:' &&
      rossrv list &&
      echo 'Available parameters:' &&
      rosparam list &&
      roslaunch  /home/nbro/Desktop/my_project/my_file.py "
ROS workspace directory: /home/nbro
Base path: /home/nbro
The specified source space "/home/nbro/src" does not exist
```
which states that the ROS workspace is at `/home/nbro`, but it is not there, it is under `/home/nbro/catwin_ws`. How can I change these configurations?  I know I can change the location of the ROS distribution (in the settings).

Of course, I can solve this issue by simply editing the run configurations in PyCharm so that it uses a Python template and not a ROS Launch file. But is it possible to change this PyCharm default behavior after having installed the ROS Support plugin, so that Python modules are executed as Python modules and not ROS Launch files?

Noam-Dori, 2019-10-15:
First, what exactly was the action you took? Did you try running something in command line or something else? The command you sent looks a bit more complicated than a simple "run" action.

For now, I am assuming the error comes out you try to run the command in the code snippet.
The entire command assumes that the ROS workspace directory is at /home/nbro. First, this is not something the plugin changes, but rather something went wrong ROS-wise. If you change in that command every `/home/nbro` to `/home/nbro/catwin_ws` this would be fixed.

nbro, 2019-10-15:
@Noam-Dori The action was: right-click on the file, then run. Yes, of course, I can run a Python module from the command-line by invoking `python my_file.py`.

> First, this is not something the plugin changes, but rather something went wrong ROS-wise. If you change in that command every /home/nbro to /home/nbro/catwin_ws this would be fixed.

I am not sure what you mean by ROS-wise. In ROS, you can create multiple workspaces, so this does not make much sense, as there should be an option to specify the workspace.

Noam-Dori, 2019-10-15:
This issue comes from [hatchery's](https://github.com/duckietown/hatchery) plugin, since they are the ones that implemented running launch files from the IDE. I don't believe this plugin overrides anything that could prevent their launch file code from working, so I think it is better if you ask them.

closed 2019-10-15 by Noam-Dori

nbro, 2019-10-15:
@Noam-Dori You're probably right. In fact, I found this issue: https://github.com/duckietown/hatchery/issues/19.