Noam-Dori/ros-integrate #18: 0.1.4 NPEs, Freezes, ISEs

Noam-Dori, 2019-10-31:
If you encountered any NullPointerExceptions, IllegalStateExceptions that were raised in the log, please report them here, along with a stack trace and the action you were doing.

If your IDE happens to freeze, please use [this guide](https://intellij-support.jetbrains.com/hc/en-us/articles/206544899-Getting-a-thread-dump-when-IDE-hangs-and-doesn-t-respond) to get a thread dump (like a stack trace, but for freezes), and if it is caused by this plugin, please report it here as well (with your thread dump attached).

Noam-Dori, 2019-11-03:
some sort of weirdness is happening that prevents a lot things in this new version:
```java
java.lang.Throwable: It's prohibited to access index during event dispatching
	at com.intellij.openapi.diagnostic.Logger.error(Logger.java:145)
	at com.intellij.openapi.project.NoAccessDuringPsiEvents.checkCallContext(NoAccessDuringPsiEvents.java:39)
	at com.intellij.util.indexing.FileBasedIndexImpl.ensureUpToDate(FileBasedIndexImpl.java:736)
	at com.intellij.util.indexing.FileBasedIndexImpl.processExceptions(FileBasedIndexImpl.java:907)
	at com.intellij.util.indexing.FileBasedIndexImpl.processValueIterator(FileBasedIndexImpl.java:983)
	at com.intellij.util.indexing.FileBasedIndexImpl.processValuesInScope(FileBasedIndexImpl.java:954)
	at com.intellij.util.indexing.FileBasedIndexImpl.processValues(FileBasedIndexImpl.java:851)
	at com.intellij.psi.search.FileNameIndexServiceImpl.getVirtualFilesByName(FileNameIndexServiceImpl.java:27)
	at com.intellij.psi.search.FilenameIndex.processFilesByName(FilenameIndex.java:87)
	at com.intellij.psi.search.FilenameIndex.processFilesByName(FilenameIndex.java:74)
	at com.intellij.psi.search.FilenameIndex.getFilesByName(FilenameIndex.java:145)
	at com.intellij.psi.search.FilenameIndex.getFilesByName(FilenameIndex.java:65)
	at ros.integrate.workspace.ROSPackageFinderBase.lambda$investigate$5(ROSPackageFinderBase.java:92)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
	at java.base/java.util.TreeMap$KeySpliterator.forEachRemaining(TreeMap.java:2739)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at ros.integrate.workspace.ROSPackageFinderBase.investigate(ROSPackageFinderBase.java:93)
	at ros.integrate.workspace.ROSPackageManagerImpl.applyChangesToOrphans(ROSPackageManagerImpl.java:118)
	at ros.integrate.workspace.ROSPackageManagerImpl.doBulkFileChangeEvents(ROSPackageManagerImpl.java:110)
	at ros.integrate.workspace.ROSPackageManagerImpl.access$300(ROSPackageManagerImpl.java:25)
	at ros.integrate.workspace.ROSPackageManagerImpl$1.after(ROSPackageManagerImpl.java:60)
	at jdk.internal.reflect.GeneratedMethodAccessor96.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeListener(MessageBusImpl.java:513)
	at com.intellij.util.messages.impl.MessageBusConnectionImpl.deliverMessage(MessageBusConnectionImpl.java:142)
	at com.intellij.util.messages.impl.MessageBusImpl.doPumpMessages(MessageBusImpl.java:438)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpWaitingBuses(MessageBusImpl.java:398)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpMessages(MessageBusImpl.java:388)
	at com.intellij.util.messages.impl.MessageBusImpl.sendMessage(MessageBusImpl.java:370)
	at com.intellij.util.messages.impl.MessageBusImpl.lambda$createTopicHandler$1(MessageBusImpl.java:241)
	at com.sun.proxy.$Proxy40.modificationCountChanged(Unknown Source)
	at com.intellij.psi.impl.PsiModificationTrackerImpl.fireEvent(PsiModificationTrackerImpl.java:70)
	at com.intellij.psi.impl.PsiModificationTrackerImpl.incCountersInner(PsiModificationTrackerImpl.java:75)
	at com.intellij.psi.impl.PsiModificationTrackerImpl.incOutOfCodeBlockModificationCounter(PsiModificationTrackerImpl.java:65)
	at com.jetbrains.cidr.lang.symbols.symtable.FileSymbolTablesCache$1.incModificationCount(FileSymbolTablesCache.java:130)
	at com.jetbrains.cidr.lang.symbols.symtable.FileSymbolTablesCache$3.done(FileSymbolTablesCache.java:205)
	at com.jetbrains.cidr.VirtualFileChangeListener.after(VirtualFileChangeListener.kt:76)
	at jdk.internal.reflect.GeneratedMethodAccessor96.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeListener(MessageBusImpl.java:513)
	at com.intellij.util.messages.impl.MessageBusConnectionImpl.deliverMessage(MessageBusConnectionImpl.java:142)
	at com.intellij.util.messages.impl.MessageBusImpl.doPumpMessages(MessageBusImpl.java:438)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpWaitingBuses(MessageBusImpl.java:398)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpMessages(MessageBusImpl.java:388)
	at com.intellij.util.messages.impl.MessageBusImpl.sendMessage(MessageBusImpl.java:370)
	at com.intellij.util.messages.impl.MessageBusImpl.lambda$createTopicHandler$1(MessageBusImpl.java:241)
	at com.sun.proxy.$Proxy21.fileWithNoDocumentChanged(Unknown Source)
	at jdk.internal.reflect.GeneratedMethodAccessor130.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90
```

Noam-Dori, 2019-11-03:
The errors above do not happen on the intelliJ platform version 2019.2.*
I specfically checked on 2019.2.3

now checking on 2019.3 EAPs

Noam-Dori, 2019-11-08:
seems like in the EAP after 193.4932 (IC-193.5096), the log error still occurs, but the indexing still properly functions. In any case, the commit above fixes the log and any possible indexing issue in the future.

peci1, 2020-02-15:
Have a look at https://youtrack.jetbrains.com/issue/CPP-19104 . I'm now pretty much convinced that this plugin is what causes the slowness and constant searching for files to index. What's more interesting - the project with which to problem manifests is not a ROS project.

Noam-Dori, 2020-02-15:
reading your .idea file and the ticket you opened, it seems you have a lot of package paths in the "additional sources" settings (40 paths to be exact), some include things that shouldn't be there like the compiled ROS path (which is already covered by ros path).
Other than the actual ROS path, I also see 5 workspaces in there.
In order to find all possible packages for use, the plugin runs a recursive filesystem lookup on all paths specified both in the "workspace" setting and the "additional package paths" setting. Seems like the plugin cannot handle indexing all of those paths.
If you empty the setting "additional package paths" comepletely, it should solve the problem. Let me know if that worked out for you!

peci1, 2020-02-15:
Hmm, what do you mean by "compiled ros path" in my ROS_PACKAGE_PATH? I don't see any build/devel paths there. 

The elevation_mapping and mapper workspaces are built with catkin tools, while the rest of the workspace is built with catkin_make. The thing with catkin_tools is that they will eventually add the path to every single package to ROS_PACKAGE_PATH. And this is something this plugin has to cope with.

Here's an example of ROS_PACKAGE_PATH from our robots, where we build the whole workspace with catkin tools:

/home/robot/workspace/cras_subt/src/tradr-resources/adaptive_traversability_joy:/home/robot/workspace/cras_subt/src/apriltag:/home/robot/workspace/cras_subt/src/apriltag_ros/apriltag_ros:/home/robot/workspace/cras_subt/src/artifact_detection:/home/robot/workspace/cras_subt/src/bag_crawler:/home/robot/workspace/cras_subt/src/camera1394:/home/robot/workspace/cras_subt/src/camera_base:/home/robot/workspace/cras_subt/src/catch_ros:/home/robot/workspace/cras_subt/src/catkin_boost_python_buildtool/catkin_boost_python_buildtool:/home/robot/workspace/cras_subt/src/catkin_simple:/home/robot/workspace/cras_subt/src/catkin_boost_python_buildtool/catkin_boost_python_buildtool_test:/home/robot/workspace/cras_subt/src/darpa_commandpost/ros_nodes/test_mapping_server/cbor:/home/robot/workspace/cras_subt/src/exploration/cloud_proc:/home/robot/workspace/cras_subt/src/control_policy:/home/robot/workspace/cras_subt/src/ros-utils/cras_cpp_common:/home/robot/workspace/cras_subt/src/ros-utils/cras_py_common:/home/robot/workspace/cras_subt/src/darpa-communications/sw-ros/darpa_carousel:/home/robot/workspace/cras_subt/src/darpa_estop_driver:/home/robot/workspace/cras_subt/src/darpa-communications/sw-ros/darpa_motes:/home/robot/workspace/cras_subt/src/darpa_msgs:/home/robot/workspace/cras_subt/src/darpa_commandpost/ros_nodes/darpa_commandpost:/home/robot/workspace/cras_subt/src/darpa_rviz_plugins:/home/robot/workspace/cras_subt/src/tradr-resources/dem_generation_msgs:/home/robot/workspace/cras_subt/src/doxygen_catkin:/home/robot/workspace/cras_subt/src/draco:/home/robot/workspace/cras_subt/src/eigen_catkin:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_extrinsic_calibration:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_gridmap_2d:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_icp_mapping:/home/robot/workspace/cras_subt/src/exploration/exploration_launchers:/home/robot/workspace/cras_subt/src/tradr-resources/exploration_msgs:/home/robot/workspace/cras_subt/src/tradr-resources/flc_msgs:/home/robot/workspace/cras_subt/src/tradr-ugv-base/flc_teleoperation:/home/robot/workspace/cras_subt/src/tradr-payload/flipper_touch_driver:/home/robot/workspace/cras_subt/src/tradr-resources/flipper_touch_driver_msgs:/home/robot/workspace/cras_subt/src/tradr-resources/gas_detector_msgs:/home/robot/workspace/cras_subt/src/tradr-resources/gas_sensor_msgs:/home/robot/workspace/cras_subt/src/tradr-payload/gas_smoke_alerts:/home/robot/workspace/cras_subt/src/gazebo_custom_sensor_preloader:/home/robot/workspace/cras_subt/src/gazebo_noisy_depth_camera:/home/robot/workspace/cras_subt/src/gazebo_rotating_lidar:/home/robot/workspace/cras_subt/src/tradr-resources/general_monitoring_msgs:/home/robot/workspace/cras_subt/src/gflags_catkin:/home/robot/workspace/cras_subt/src/glog_catkin:/home/robot/workspace/cras_subt/src/eigen_checks:/home/robot/workspace/cras_subt/src/tradr-resources/gpio_teleop_joy:/home/robot/workspace/cras_subt/src/tradr-ugv-base/gps_from_odom_updater:/home/robot/workspace/cras_subt/src/tradr-resources/gps_from_odom_updater_msgs:/home/robot/workspace/cras_subt/src/tradr-resources/jetson_media:/home/robot/workspace/cras_subt/src/darpa_commandpost/ros_nodes/test_mapping_server/json:/home/robot/workspace/cras_subt/src/libnabo:/home/robot/workspace/cras_subt/src/libpointmatcher:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/libpointmatcher_ros:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_icp_mapper:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_icp_mapper_experiments:/home/robot/workspace/cras_subt/src/ethzasl_icp_mapping/ethzasl_point_cloud_vtk_tools:/home/robot/workspace/cras_subt/src/tradr-resources/librover_common_catkin:/home/robot/workspace/cras_subt/src/lms1xx:/home/robot/workspace/cras_subt/src/darpa_commandpost/ros_nodes/test_mapping_server/mapping_relay:/home/robot/workspace/cras_subt/src/darpa_commandpost/ros_nodes/test_mapping_server/mapping_server:/home/robot/workspace/cras_subt/src/metis_catkin:/home/robot/workspace/cras_subt/src/gtsam_catkin:/home/robot/workspace/cras_subt/src/mikrotik_swos_tools:/home/robot/workspace/cras_subt/src/minkindr/minkindr:/home/robot/workspace/cras_subt/src/minkindr_ros/minkindr_conversions:/home/robot/workspace/cras_subt/src/minkindr_gtsam/minkindr_gtsam:/home/robot/workspace/cras_subt/src/mincurves/mincurves:/home/robot/workspace/cras_subt/src/tradr-laser-slam/laser_slam:/home/robot/workspace/cras_subt/src/tradr-laser-slam/laser_slam_ros/laser_mapper:/home/robot/workspace/cras_subt/src/tradr-resources/mobilicom_media:/home/robot/workspace/cras_subt/src/mrs_msgs:/home/robot/workspace/cras_subt/src/mrs_lib:/home/robot/workspace/cras_subt/src/detection_localize:/home/robot/workspace/cras_subt/src/msg_db:/home/robot/workspace/cras_subt/src/tradr-resources/networkanalysis_msgs:/home/robot/workspace/cras_subt/src/tradr-ugv-base/nifti_drivers_launchers:/home/robot/workspace/cras_subt/src/tradr-u [... cut ...] c_map_base:/home/robot/workspace/cras_subt/src/volumetric_mapping/volumetric_mapping:/home/robot/workspace/cras_subt/src/volumetric_mapping/volumetric_msgs:/home/robot/workspace/cras_subt/src/volumetric_mapping/octomap_world:/home/robot/workspace/cras_subt/src/tradr-laser-slam/laser_slam_ros/laser_mapper_tools:/home/robot/workspace/cras_subt/src/voxel_map:/home/robot/workspace/cras_subt/src/wifi_scan:/home/robot/workspace/cras_subt/src/tradr-resources/wireless_network_msgs:/opt/ros/melodic/share

peci1, 2020-02-15:
And I can confirm that reducing the additional packages path to the 5 real workspaces helped.

However, I don't get suggestions in .msg files for anything else than the current package (e.g. even std_msgs aren't suggested).

peci1, 2020-02-15:
What you could do to automatically reduce the package path length is searching for files like `.catkin_workspace` in each path of ROS_PACKAGE_PATH, and only add directories containing these special files.

Noam-Dori, 2020-02-16:
> what do you mean by "compiled ros path" in my `ROS_PACKAGE_PATH`?

When I say "compiled ros path" I refer to `/opt/ros/melodic`, which has all the binary (pre-compiled) packages you install on your platform. This path is covered by the setting "[ROS Path](https://github.com/Noam-Dori/ros-integrate/wiki/The-Settings-Page#ros-path)"

> The thing with catkin_tools is that they will eventually add the path to every single package to ROS_PACKAGE_PATH. And this is something this plugin has to cope with.
Here's an example of ROS_PACKAGE_PATH from our robots, where we build the whole workspace with catkin tools: ...

Good to know especially since I am not familiar with the newer build systems (e.g catkin_tools, colcon_build, etc.)
However, the [ROS wiki](http://wiki.ros.org/ROS/EnvironmentVariables) points out that since the introduction of catkin there is no real need to use `ROS_PACKAGE_PATH`:

_"ROS_PACKAGE_PATH is an optional, but very common environment variable that allows you to add more ROS packages from source to your environment.
...
With the introduction of catkin, the need to manually update ROS_PACKAGE_PATH becomes obsolete, and is only necessary for backwards compatibility with rosbuild packages. "_
Moreover, it should be used to add packages _outside_ the workspace.

> Here's an example of ROS_PACKAGE_PATH from our robots ...

Another suggestion I have for you to improve your experience with the plugin is to put the path to the robot's workspace (`/home/robot/workspace/cras_subt`) in the "[workspace path](https://github.com/Noam-Dori/ros-integrate/wiki/The-Settings-Page#workspace-path)" setting. That path is used to find all packages in your workspace without resorting to `ROS_PACKAGE_PATH`. Moreover, since it is unique per-project, it allows you to switch between workspaces without reconfiguring the workspace.

> I don't get suggestions in .msg files for anything else than the current package (e.g. even std_msgs aren't suggested).

This is a known bug that will hopefully be solved before version 0.2. Please report anything about the exact steps you took (settings, etc.) to get this bug in the [issue's ticket](https://github.com/Noam-Dori/ros-integrate/issues/23).

> What you could do to automatically reduce the package path length is searching for files like .catkin_workspace in each path of ROS_PACKAGE_PATH, and only add directories containing these special files.

Sadly I cannot do that. `ROS_PACKAGE_PATH` can also direct to packages that are not inside any workspace. Moreover, in the example you provided, the plugin would need to check the parent of these paths rather than those paths themselves for a `.catkin_workspace` file. It would also not remove any of the effort because these paths are still searched recursively.

peci1, 2020-02-16:
Thanks for the insights. The I'd say it'd be nice if the plugin issued a warning in case it finds /opt/ros/*/setup.bash in the workspaces path. It could also issue a warning when the ROS_PACKAGE_PATH is long, with a suggestion what the user should do (point to the toplevel directory).

As ROS_PACKAGE_PATH is obsolete, shouldn't you maybe look at CMAKE_PREFIX_PATH? I think that's the one that is searched by catkin.

And, in the end, do you say that with the multitude of packages I have, this plugin will be slow regardless of the settings?

Noam-Dori, 2020-02-16:
> As ROS_PACKAGE_PATH is obsolete, shouldn't you maybe look at CMAKE_PREFIX_PATH? I think that's the one that is searched by catkin.

`CMAKE_PREFIX_PATH` only points to the ros path, not the workspace. catkin actually does no work figuring out which workspace is the right one, it either:
1. catkin_make: it runs a special cmake file it searches for in the working directory (which is why it requires you to run catkin_make from workspace/src) which runs a recursive search for all other packages and loads them.
2. catkin_make_isolated: it runs a special _python_ script which runs a recursive search for all other packages that are children of the current working directory (which is why workspace/src is required once again) and loads their cmake files.

> The I'd say it'd be nice if the plugin issued a warning in case it finds /opt/ros/*/setup.bash in the workspaces path. ...

Feel free to open a feature ticket with more details. 

> And, in the end, do you say that with the multitude of packages I have, this plugin will be slow regardless of the settings?

The recursive part of the indexing stops once a package is found, so it take around as long as a catkin_make takes to load the workspace. Only one way to truly find out.

Cheers!

peci1, 2020-02-16:
What you are referring to is the catkin build tool. However, there is also the catkin python module along with rospack, which allow you to search for packages during runtime. These have to take some env vars into accout, as they cannot rely on the current directory...

closed 2020-03-07 by Noam-Dori