Noam-Dori/ros-integrate #21: Can you add a Wiki/tutorial to setup a project ?

remi2257, 2020-01-08:
Hi, 

I'm trying to use your plugin to add rospy to my work environment but it's kinda hard to figure out what to do with it ^^ It's basically the same issue as #3 

When I add it to PyCharm, it leads to an inspection of my home directory, which is not where my ROS package is located, and which also leads to "every line of your code is duplicated from another source" because, my repo is basically scans 2 times.

I think that I have to add some paths in .xml files from the .idea folder but I do not know which field I should fill ... :stuck_out_tongue: 

Can you help on that ? Maybe, you could do a tutorial ?

Cheers

Noam-Dori, 2020-01-08:
First, could you add a screenshot of the false warnings and what sort of files were scanned incorrectly? It could help find the problem.
The only duplication inspections I currently have are about .msg files, since only 0.1.5 starts dealing with package.xml files so I suspect this may be a PyCharm thing.
Like I asked on issue #3, what would you want to see in a tutorial/wiki? At the moment the project setup is supposed to be completely automatic, though it can be manually configured from the settings page (under languages & frameworks in the settings menu).

remi2257, 2020-01-09:
Hey,

I do not want to rerun the inspect, it took way to long, but basically, as it runs from "~/" path, it will soon or later run into my project and re-inspect it.

Currently, I have none automatic configuration at all. Moreover, as it runs from USER_HOME_PATH, it won't find ROS folder because it's installed in _/opt/ros/melodic/_

This issue will be fixed with my second point, do not worry about it :smile: 

---

> Like I asked on issue #3, what would you want to see in a tutorial/wiki? At the moment the project setup is supposed to be completely automatic, though it can be manually configured from the settings page (under languages & frameworks in the settings menu).

Ok, that a first good point, I did not know that :bulb:

You should mentioned that on your PyCharm documentation. Information are lost among the history Version, while it could be so simple to just mentioned it.  

Like : 

ROS Support Plugin come up with 3 path values:
1) ROS Path : it is filled automatically with $ROS_ROOT value. This could be done by adding the line _export ROS_ROOT=/opt/ros/melodic_ to your .bashrc file
2) Workspace path : ...
3) Additionnal package path (optinional) :  ....

I think I got the first one, but I do not understand what should be write on workspace path field ?

Cheers

Noam-Dori, 2020-01-09:
First, thanks for pointing out what information you would like to see. This can help me know what I need to put in the wiki.

> You should mentioned that on your PyCharm documentation. Information are lost among the history Version, while it could be so simple to just mentioned it.

The documentation from the changelogs will be copied to the wiki. Any specific details you are intrested in?

> I think I got the first one, but I do not understand what should be write on workspace path field ?

I will write this in the wiki, but this is the concept behind all 3 (and soon more) options in the dialog:
1. ROS Path: as you guessed correctly, this is the path to the root of the installed ROS directory. This path is usually `/opt/ros/<version>` but people have different settings and it could be installed elsewhere.
2. Workspace path: this is the path to your catkin workspace, the root folder containing all ros packages, build files, etc. It directly contains a file called `.catkin_workspace`.
3. Additional Package Paths: This is a colon-separated list of paths to all ROS packages outside of the workspace (not recommended ROS practice, but available nonetheless).

> Currently, I have none automatic configuration at all.

Here are potential reasons why the settings variables were not set for you:
1. ROS Path uses `$ROS_ROOT`. The two are not the same: ROS Path is the grandparent of the ROS root, which points to the cmake of the core ros package. this value is set in `.bashrc` with the command `source /opt/ros/<version>/setup.bash`. Do you run this command in your .bashrc before loading PyCharm? without it, the IDE cannot know where anything ROS is located.
2. Workspace path does not use any environment variable and searches for a folder that contains a .catkin_workspace file. Does your project contain such a file? If not, the plugin will try to guess where it is (and potentialy fail)
3. Additional package paths is exactly $ADDITIONAL_PACKAGE_PATH. This is completely optional since the introduction of catkin, so it can be left empty.

remi2257, 2020-01-09:
Thank you for your answer, you detailed well and I think it will be a very good start for your Wiki :smile: 

I'm starting to have a good handle of ROS on PyCharm ! 

I have another question but I do not want to create a new issue because it might fit also on the Wiki -> Different imports work well for .py files, but PyCharm still doesn't like things like : 
`from my_pckg_msgs.srv import ObjDetection`
I got the same problem with messages even if the version history mentionned :
> added support for *.srv files

So, is it supposed to work, and if so, how can I make it works :) ?
(To answer your first question, Yes, it is working with a rosrun command :yum: )

Cheers and thanks again !

Noam-Dori, 2020-01-09:
This issue usually happens when the IDE is not set up using the commandline.
[This page](http://wiki.ros.org/IDEs#PyCharm_.28community_edition.29) discusses how to set up PyCharm for ROS, which is general setup that is required for now with or without the plugin.
However, from what you posted here it seems you already did that, so I am not sure.
My plugin has a long way to go before it can support messages/services in python/c++ files but it is definatly planned. At the moment, it only highlights the files, adds inspections, and adds autocompletion to them (with references to other .msg files).

Cheers!

closed 2020-01-09 by Noam-Dori

remi2257, 2020-01-10:
Hey,

Just to let you know that this trick didn't work for referencing srv & msg files, but that's not a big deal.

Do not hesitate to notify me if new updates come :smile: 

Thank you once again, pleasure to work with you

Rémi