Noam-Dori/ros-integrate #30: Avoid whole computer indexing

remi2257, 2020-03-16:
**Feature:**
I would like PyCharm to not index files when Workspace (Langugages & Framworks > ROS) is not filled

**Background**
The plugin always want to index. It starts to index my whole computer as soon as I create a project because initially, the workspace is empty which corresponds to the PC's root (I guess)
As it is not possible to enable/disable plugins for one specific project, ROS Support is always enable on my PyCharm and each time I create a new project, my computer almost freeze because PyCharm is indexing everything

**Details**
Can you add an option/condition which will allow set a new external library only if the user set one :) ?

Cheers

Noam-Dori, 2020-03-18:
fixed

closed 2020-03-18 by Noam-Dori

remi2257, 2020-03-30:
Thank you sir :)