Noam-Dori/ros-integrate #38: package format 2?

peci1, 2020-04-06:
**Feature:**
Support package.xml format 2.

**Background**
I've some format 2 packages, and any time I want package suggestions in `<depend>` or `<exec_depend>` tags, CLion tells there is nothing to suggest. `<build_depend>` works as it is also a format 1 tag.

**Details**
Version 0.1.5., CLion 2020.1 Beta

Noam-Dori, 2020-04-06:
When you restart CLion, does it temporarily solve the issue?
If so, this might be caused by bug #23 where the component responsible for finding packages randomly clears its cache.
If this is the case, the bug is fixed in the next version, 0.1.6, though you can also get the fix right now if you install the plugin from source.

peci1, 2020-04-06:
No, restarting does not help. I did the test right after starting CLion, as soon as the project was loaded.

peci1, 2020-04-06:
Ah, disabling Hatchery fixes this. Is there a way to get inspections from both?

Noam-Dori, 2020-04-06:
I think the reason hatchery interferes with this plugin's inspections is that it converts the file type of package.xml files to its own special files since this plugin does not change anything about the physical package.xml file.
I think you could raise the issue to hatchery to make their special treatment of package.xml files less interfering.

peci1, 2020-04-06:
Maybe it'd be more efficient if you talked to hatchery guys directly? I'm not sure I'd be of much help in resolving this issue...

Noam-Dori, 2020-04-06:
I opened an issue on their github page:
https://github.com/duckietown/hatchery/issues/30
Hopefully this issue could be resolved

closed 2020-04-06 by Noam-Dori

peci1, 2020-04-06:
Great, thanks!