Noam-Dori/ros-integrate #39: Somehow interferes with CMake project loading.

peci1, 2020-04-06:
**Describe the bug**
After CLion is started and loading a previous project, all project files are reported as not belonging to the project.

**To Reproduce**
1. Have a CMake C++ project, set up the ROS plugin for it.
1. Exit CLion
1. Start clion
1. load the previous project

**Expected behavior**
The project files should be recognized as project files. Sometimes, the non-recognized project files get a weird yellowis-brownish background in the project view, sometimes they have normal background. When I press the "reload cmake project" button, project files are recognized correctly.

**Environment Information:**
 - OS: *Ubuntu 18.04*
 - IDE: *CLion 2020.1 beta* (but has been happening all the time)
 - Plugin Version: *0.1.5* (but has been happening all the time)
 - Related Plugins: *hatchery*

**Additional context**
*Add any other context about the problem here.*

Noam-Dori, 2020-04-06:
Could you send a screenshot of the bugged project view? Also, what are your ROS settings? more specifically, the three "Environment" paths (ROS Path, Workspace, Additional Package Paths)?

peci1, 2020-04-06:
[image]
I've also opened the CMake window so that you can see what happened during the project load. Nothing interesting seems to be there.

*ROS Path*: /opt/ros/melodic
*Workspace*: /media/data/subt/thirdparty/ws
*Additional package path*: empty

Here's CLion console output: [clion.txt](https://github.com/Noam-Dori/ros-integrate/files/4439933/clion.txt). I see there's a lot of hatchery-related bugs, but I thought this plugin is aware of hatchery and tries to cooperate...

Noam-Dori, 2020-04-06:
Are you talking about the warning war above the file that says 
"This file does not belong to any project target, code insight features might not work properly?"

peci1, 2020-04-06:
Yes, this warning. CLion is then unable to suggest project-related inspections and only suggests syntax-based things.

Noam-Dori, 2020-04-06:
This is actually an extension of a CLion bug:
https://youtrack.jetbrains.com/issue/CPP-10204
The root of the issue is that CLion uses the CMake load action to index information used for most of its code insight features, and thus requires loading the CMake project every time you open CLion. I think this can be solved by turning on auto-reload in `Build, Execution, Deployment > CMake` but I am not too sure.

peci1, 2020-04-06:
I tested the plugin with Hatchery disabled. The bug is still there. I reloaded the project, quit clion, started it again, and project files are not recognized.

When I disable ROS integrate and enable Hatchery, the bug is not there.

Auto-reloading CMake project is not an option for me, that would eat up my CPU...

peci1, 2020-04-06:
Are you able to replicate this issue?

Noam-Dori, 2020-04-06:
I stand corrected. Will get to working on a fix.

Noam-Dori, 2020-04-25:
The issue seems to be fixed, and is available on a separate branch for now. However:
1. this was tested in a WSL, not directly on Ubuntu.
2. In order for the fix to work for all IntelliJ-based IDEs, I created a new module type. This might interfere with the functionality of the original modules of the IDEs in ways I am not aware of, so any tests on actual workspaces in CLion, IntelliJ IDEA, and PyCharm would be greatly appreciated.
It is for these reasons that I posted this fix in a seperate branch, f-stable-model (see #41) and added a specific pull request to see if any more problems are raised.

peci1, 2020-04-25:
Great. How can I test it?

Noam-Dori, 2020-04-25:
First clone this repository somewhere.
then open the commandline of your system and change directory to the root of this repository (ros-integrate)
If you are on windows:
run `gradlew buildPlugin`
If you are on linux:
run `./gradlew buildPlugin`
this should create a new library in `build/distributions` named `ros-integrate-0.1.6.zip`
Next, open your IDE and go to the "plugins" tab in the IDE/project settings.
in the top you should see a gear icon next to the Marketplace and Installed tab names.
Click on that gear and select the "install plugin from disk" option.
in the menu that opens select the `ros-integrate-0.1.6.zip` library you just created.
This should enable the new development version of the plugin.
If you cannot find the .zip library, there is another library in `build/libs` named `ros-intrgrate-0.1.6.jar` that could work as well.

peci1, 2020-04-25:
Gradlew doesn't work for me because this repo is missing `gradle/wrapper/gradle-wrapper.jar` file. I tried with system gradle, but:

    > Task :compileJava 
    /tmp/ros-integrate/src/main/java/ros/integrate/pkg/ROSWorkspacePackageFinder.java:33: error: reference to Module is ambiguous
         private Module loadedModule = null;

Could you maybe provide the built zip instead?

Noam-Dori, 2020-04-25:
the built zip is available here:
https://github.com/Noam-Dori/ros-integrate/releases/tag/v0.1.6b1
I also updated the branch with the gradle wrappers if you still wish to compile from source.

closed 2020-08-25 by Noam-Dori