Noam-Dori/ros-integrate #44: Uninstall plugins don't work well.

gachiemchiep, 2020-06-20:
Recently, I don't use ROS anymore, so I uninstall this plugin. 
Unfortunately, the ROS directory is still left inside "External Libraries". As in the below image

[image]

I tried everything, include uninstall and removing all of pycharm's config, log, cache. But it didn't work well. 

```bash
~/.config/JetBrains/PyCharm2020.1
~/.cache/JetBrains/PyCharm2020.1
~/.local/share/JetBrains/PyCharm2020.1
~/.cache/JetBrains/PyCharm2020.1/log
```

Do you know how to solve this?

Noam-Dori, 2020-06-20:
find your project's .idea folder. There should be two files you need to modify:

- first, your module's .iml file. In there you should see an XML tag called \<module>...\</module>
in it, you will see one or two \<orderEntry/> tags you might need to delete. their type is library, and their names are "ROS" and "workspace". deleting the tags will remove the dependecy to those libraries and remove them visualy.
- second, and last, is the library file. In some versions, libraries were stored in a separate folder called libraries. In other versions of the IDEA framework, the library files are stored directly in the .idea folder. Either way, delete the files "ROS.xml" and "workspace.xml" to remove the libraries completely.
- last, since you are uninstalling the plugin, you can also delete the plugin's settings file, stored directly in the .idea folder, called "ros.xml". Deleting it would remove any trace of those settings.

gachiemchiep, 2020-06-20:
Hello @Noam-Dori 
Thanks for your advice. 
I delete the .idea directory and everything works fine for now.

I already gave you 5 stars on the marketplace. 
I hope that your plugin will get more popular in the future.

closed 2020-06-20 by gachiemchiep