Noam-Dori/ros-integrate #47: Distribution dependencies not resolved properly

Doomerdinger, 2020-08-27:
**Describe the bug**
Packages found in distribution ROSDep source lists are not properly found. Autocomplete does not function on them either.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a valid package.xml file
2. Observe that standard ros packages (such as roscpp or catkin) are not found properly, indicating an unresolved dependency.

**Expected behavior**
Dependency is properly found, and autocomplete works just as it would on packages found in the base.yaml dependency list.

**Environment Information:**
 - OS: Ubuntu 18.04
 - IDE: CLion 2020.2.1
 - Plugin Version: 0.15 (and the current master 0.16)
 - Related Plugins: None

**Stack Trace**
*If the bug causes an error or an IDE freeze, [get its thread dump](https://intellij-support.jetbrains.com/hc/en-us/articles/206544899-Getting-a-thread-dump-when-IDE-hangs-and-doesn-t-respond) and paste it here.*

**Screenshots**
[image]

**Additional context**
Issue persists even if the distribution is added directly.
To be more clear, rosdep itself gets the location from [index.yaml](https://github.com/ros/rosdistro/blob/master/index.yaml) but assuming this plugin cannot do that it does not work even when adding the distribution itself (in my case https://raw.githubusercontent.com/ros/rosdistro/master/melodic/distribution.yaml) to the "Online ROSDep Source Lists".

Noam-Dori, 2020-08-28:
This is actually not a bug with rosdep, but a bug with the compiled packages finder.
Related issues/PRs:
- #39
- #41
As of now, the issue was solved for older versions of CLion with the current commits, but not the current version.

I provided a more technical explanation of the bug [here](https://github.com/Noam-Dori/ros-integrate/pull/48#issuecomment-682840349)

Noam-Dori, 2020-09-17:
Small technical update: The issue is probably caused because the plugin is failed to update components properly.
What I did was convert the package manager to a "Project Service" whcih does not work on startup... It only works when explicitly requested, which may break things...
What I should've done is convert it two pieces: a "project startup activity" which boots all the library things, and a project service which takes care of package indexing and retrieval.

So far, I created a test class for the library indexing (this part is what currently causes the bug):
```java
public class AbridgedLibrary implements StartupActivity {
    @Override
    public void runActivity(@NotNull Project project) {
        ApplicationManager.getApplication().invokeLater(() -> {
            Library lib = WriteCommandAction.runWriteCommandAction(project,
                    (Computable<Library>) () -> loadLibrary(project));
            Arrays.stream(ModuleManager.getInstance(project).getModules())
                    .forEach(module -> setDependency(module, lib));
        });
    }

    private @NotNull Library loadLibrary(Project project) {
        LibraryTable table = LibraryTablesRegistrar.getInstance().getLibraryTable(project);
        Library lib = table.getLibraryByName("my_lib");
        if (lib == null) {
            lib = table.createLibrary("my_lib");
            String path = "C:\\my\\path\\to\\ros\\kinetic";
            Library.ModifiableModel model = lib.getModifiableModel();
            String url = VirtualFileManager.constructUrl(LocalFileSystem.PROTOCOL, path);
            VirtualFile file = VirtualFileManager.getInstance().findFileByUrl(url);
            if (file != null) {
                model.addRoot(file, OrderRootType.CLASSES);
                model.commit();
            }
        }
        return lib;
    }

    private void setDependency(Module module, Library lib) {
        ModifiableRootModel model = ModuleRootManager.getInstance(module).getModifiableModel();
        LibraryOrderEntry entry = model.findLibraryOrderEntry(lib);
        if (entry == null) {
            ModuleRootModificationUtil.addDependency(module, lib);
        }
        model.dispose();
    }
}
```

I also tested it on:
- IntelliJ IDEA 2020.2 on Windows
- CLion 2020.2 on Windows

So far, seems promising. Working now on figuring out how exactly I want to apply this change on the actual component

Noam-Dori, 2020-09-20:
Extra details on this commit:
I tested this commit on the following platforms:
- CLion 2020.2 (Windows) both in workspace scale project and single-package project
- IntelliJ IDEA 2020.2 (Windows) only in workspace scale project
- CLion 2019.3 (Windows) only in workspace scale project
- IntelliJ IDEA 2019.3 (Windows) only in workspace scale project

In CLion 2020.2 I found that the "Relead CMake" event completely erases the dependencies of the CMake module to the ROS libraries. This probably also happens in CLion 2019.3
To fix this, I need to subscribe to the CMake actions topic (`CMakeWorkspaceListener`) and trigger dependency reload manually. However, the topic only exists in CLion code, not in IntelliJ SDK code.
I'm working on figuring out a way to do some level of "conditional compilation" so I don't have to publish two separate plugins

Noam-Dori, 2020-09-22:
Update:
after doing some testing, I added a [pre-release](https://github.com/Noam-Dori/ros-integrate/releases/tag/v0.1.6b3) with the plugin binaries to test out.

It is currently untested on Ubuntu, so feedback for testing on Ubuntu will help out a lot.
This release should fix both this issue and #46

Noam-Dori, 2020-10-05:
This issue seems to be solved with the latest commits. These commits have been up as binaries and source code for about two weeks, so I assume the inactivity on this issue means the commits worked.

I also finalized the version and uploaded it to the plugin repository
If you still see this problem in the plugin, please open a new issue

closed 2020-10-05 by Noam-Dori