Noam-Dori/ros-integrate #52: Features Currently Supported

wmmc88, 2020-10-26:
* Is there somewhere that you list all the currently supported features(ex. what filetypes there's parsing for, what ros things the plugins support, etc.)?

* Does this plugin currently support the build process or is that something that needs to be done outside of Clion? If it does, is it compatible with `catkin_tools` or is it only compatible with `catkin_make`?

* Have you seen https://github.com/paoloach/ROS-JetBrains-Plugin and https://github.com/duckietown/hatchery? How will this plugin differ from those? What featureset will it have compared to those? 

**Plugin Version**
0.1.6

Noam-Dori, 2020-10-26:
Hi @wmmc88 , 
> Is there somewhere that you list all the currently supported features?

If you are looking for a technical feature list, you can see it in the changelog, either on the [JetBrains Plugin Repository](https://plugins.jetbrains.com/plugin/11235-ros-support/versions) under "What's New" or as the [raw changelog file](https://github.com/Noam-Dori/ros-integrate/blob/master/changelog.html).

Less technical documentation is available on the [wiki](https://github.com/Noam-Dori/ros-integrate/wiki) but there is a lot missing from it. If you need documentation on a specific feature or concept, let me know.

> what filetypes there's parsing for

Currently, the following files get extra functionality by the plugin:
- "Message files" `.msg`,`.srv`,`.action`
- "Manifest file" `package.xml`
- Not really a file, but you can see conditions in `.condition` files.

> what ros things the plugins support

I suggest having a look at [this wiki page](https://github.com/Noam-Dori/ros-integrate/wiki/Basic-Features), it contains a somewhat simplified feature list of the ROS things you get support for. Again, a more comprehensive list can be found in the changelog.

> Does this plugin currently support the build process

Currently the plugin does not support custom build processes

>  or is that something that needs to be done outside of Clion?

CLion natively supports CMake, so it supports some aspect of the ROS build process on its own. You can load one package at a time by loading a specific CMakeLists.txt file or load several by having a more top-level CMakeLists.txt file that has `add_subdirectory` to each package you want to index. The normal top-level CMakeLists.txt found in ROS1 catkin workspaces does this job too, though it does not do isolation

> If it does, is it compatible with `catkin_tools` or is it only compatible with `catkin_make`?

From my tests, only `catkin_make` (not even `catkin_make_isolated`) fully work within the context of CLion since they run multiple CMake operations.
However, you can use the "second toplevel" strategy I showed above to basically have two indexing strategies. This is how I managed to work with isolation. It could be supported in the future, however. This is something I plan to tackle in the future.

> Have you seen https://github.com/paoloach/ROS-JetBrains-Plugin and https://github.com/duckietown/hatchery? How will this plugin differ from those? What featureset will it have compared to those?

I don't know exactly what features those plugins offer, but the idea is that my plugin will cover features the users of this plugin need for ROS development.
For example, you asked a question about a list of features, so I added a new wiki page with a list of features.
Therefore, the difference between this plugin to other plugins is up to those who post issues here.

wmmc88, 2020-10-26:
Thanks for making me aware of the wiki! Answered a lot of my questions.

Does this plugin require that I launch clion from a shell where i've already sourced `/opt/ros/<ros_version>'? And if it is, is this something the plugin could somehow do?

[This](https://github.com/Noam-Dori/ros-integrate/wiki/The-Settings-Page) page says that you use the `.catkin_workspace` file to find the path of the catkin workspace. this only works in catkin_make environments. In `catkin_tools` (what most users,  companies, major ros orgs use), there is a `.catkin_tools` directory instead. 

I added a fake top level cmake lists to include subdirectories of my packages/metapackages. Is it possible to have the plugin generate and maintain this behind the scenes?(feature suggestion)

I'm also not sure if I misconfigured your plugin because I don't think functionality is working for me. I can see things like .msg files getting a custom icon, but if I CTRL+Click a license(MIT) in a license tag in a package xml, I just get a "Cannot find declaration to go to". I see the ros settings page and have set the correct ros path and workspace.

Noam-Dori, 2020-10-26:
> Does this plugin require that I launch clion from a shell where i've already sourced `/opt/ros/<ros_version>`?

The plugin does not require you to start CLion from source, CLion technically requires that. The plugin will work fine without sourcing, though some of the auto-detection is lost, meaning you need to put in the "ros path" on your own.
CLion needs the sourcing since the sourcing changes CMAKE_PREFIX_PATH. You can add it manually in the CMake settings and it should work fine.

> And if it is, is this something the plugin could somehow do?

This is something I am planning on implementing

> This page says that you use the `.catkin_workspace` file to find the path of the catkin workspace. this only works in `catkin_make` environments. In `catkin_tools` (what most users, companies, major ros orgs use), there is a `.catkin_tools` directory instead.

.catkin_workspace is just a simple placeholder that tells the plugin that this is the root workspace. The plugin only uses this for detecting the root directory. As far as I am aware, .catkin_tools does the same thing. It should not be hard to add .catkin_tools to the detector. Feel free to add a [feature request](https://github.com/Noam-Dori/ros-integrate/issues/new?assignees=Noam-Dori&labels=feature&template=feature_request.md&title=) if you would like this feature

> I added a fake top level cmake lists to include subdirectories of my packages/metapackages. Is it possible to have the plugin generate and maintain this behind the scenes?(feature suggestion)

Again, this is something I want to implement in the future.

> I'm also not sure if I misconfigured your plugin because I don't think functionality is working for me. I can see things like .msg files getting a custom icon, but if I CTRL+Click a license(MIT) in [...]

If you are Ctrl+Clicking on a license value, your configurations should not matter. I also need more details, like the plugin version, and possibly a screenshot. I reccomend opening a [bug report](https://github.com/Noam-Dori/ros-integrate/issues/new?assignees=Noam-Dori&labels=bug&template=bug_report.md&title=) so we can give this problem more focus.

> Happy to see someone actively working on better ros support for clion! 😀

And I am happy to see people actively interacting with it 🙂

closed 2020-10-26 by wmmc88