Noam-Dori/ros-integrate #53: License feature doesnt work in CLion

wmmc88, 2020-10-26:
**Describe the bug**
https://github.com/Noam-Dori/ros-integrate/issues/52#issuecomment-716621973
After installing the latest version of the plugin available on Clion, opening a package.xml and ctrl+clicking a license value does not do anything. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to a package.xml
2. Ctrl+Click a license

**Expected behavior**
I believe its suppposed to give me a tldr of the license or open a browser to a copy of that license

**Environment Information:**
 - OS: Ubuntu 20.04.1
 - IDE: CLion 2020.2.4
 - Plugin Version: 0.1.6
 - Related Plugins: No other plugins are installed currently

**Stack Trace**
N/A

**Screenshots**
[image]
[image]

**Additional context**
N/A

Noam-Dori, 2021-07-28:
Looking at the screenshot, the issue is caused by the plugin not recognizing `uwrt_mars_rover_hw` as a ROS package (otherwise in the project view it will be seen as a blue folder).

I could not replicate the bug on my end. However, I do have several questions that can help resolve the issue:
- are the packages correctly found after restarting the IDE, or does the plugin only find random (but not all) packages?
- In the external libraries tab, is there a library called ROS? are the packages in it colored orange?

Related issues: PR #41

Noam-Dori, 2022-08-03:
Closed due to inactivity.

closed 2022-08-03 by Noam-Dori