Noam-Dori/ros-integrate #58: ArrayIndexOutOfBoundsException

SimonIT, 2020-11-11:
The error just appeared after installing

**Environment Information:**
PyCharm 2020.2.3 (Professional Edition)
Build #PY-202.7660.27, built on October 6, 2020
For educational use only.
Runtime version: 11.0.8+10-b944.34 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Linux 5.4.0-52-generic
GC: ParNew, ConcurrentMarkSweep
Memory: 1927M
Cores: 8
Non-Bundled Plugins: CMD Support, com.alayouni.ansiHighlight, com.intellij.ideolog, com.almightyalpaca.intellij.plugins.discord, mobi.hsz.idea.gitignore, org.toml.lang, com.jetbrains.intellij.datalore, ru.meanmail.plugin.requirements, ros-integrate
Current Desktop: ubuntu:GNOME
Ubuntu 18.04.5 LTS
Ros Melodic

**Stack Trace**
```
java.lang.ArrayIndexOutOfBoundsException: Index 0 out of bounds for length 0
	at ros.integrate.pkg.ROSCompiledPackageFinder.getROSRoot(ROSCompiledPackageFinder.java:49)
	at ros.integrate.pkg.ROSCompiledPackageFinder.inFinder(ROSCompiledPackageFinder.java:134)
	at ros.integrate.pkg.ROSPackageFinderBase.lambda$investigate$1(ROSPackageFinderBase.java:98)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:176)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at ros.integrate.pkg.ROSPackageFinderBase.investigate(ROSPackageFinderBase.java:101)
	at ros.integrate.pkg.ROSPackageManagerImpl.lambda$applyChangesToOrphans$3(ROSPackageManagerImpl.java:96)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:201)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:802)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$invokeLater$4(ApplicationImpl.java:322)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:84)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:132)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:47)
	at com.intellij.openapi.application.impl.FlushQueue$FlushNow.run(FlushQueue.java:188)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:971)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:841)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:452)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:744)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$9(IdeEventQueue.java:451)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:802)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:499)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

Noam-Dori, 2020-11-12:
Hi Simon,
This error is a bug caused by not having a ROS path (in the settings). It should not impact anything.
You can remove this error for now by filling in the path to your ROS distribution (in your case `/opt/ros/melodic`) in the ROS settings.
This error will not pop up in future versions at all as it has been fixed (issue #49 dealt with it) 

I hope this helps deal with the error popup

SimonIT, 2020-11-12:
Thanks for the reply!
I'm an absolute newbie to ros. Where do I find the ROS settings?

closed 2020-11-12 by SimonIT

Noam-Dori, 2020-11-12:
You can find the ROS settings in `Settings->Languages & Frameworks->ROS`
You can find detailed documentation on the settings page [on the wiki](https://github.com/Noam-Dori/ros-integrate/wiki/The-Settings-Page)