Noam-Dori/ros-integrate #60: Install-space workspaces are wrongly process from Additional package paths

peci1, 2021-03-03:
**To Reproduce**
Steps to reproduce the behavior:
1. Create a workspace with some packages, `catkin_make install` it (in my case, `/media/data/subt/virtual`).
2. Create another workspace that depends on the previous one.
3. Open a package from the second WS in CLion.
4. Configure additional package path from ROS_PACKAGE_PATH (the installed WS will be mentioned as `/media/data/subt/virtual/install/share`).
5. Open package.xml from the already open project and add a dependency on some package from the first WS.
6. The dependency will be reported as missing.
7. If I manually edit the Additional package path and remove the `/share` bit from the installed WS path, resolving its packages works.

**Expected behavior**
Packages from installed WSs should work out of the box.

**Environment Information:**
 - OS: 18.04
 - IDE: 2021.1 EAP
 - Plugin Version: master branch
 - Related Plugins: no Hatchery

Noam-Dori, 2021-07-26:
I replicated the bug. It applies not just to installed workspaces, but to 2nd level directories as well.
In the meantime, the issue seems to be circumvented either by restarting the IDE after saving the settings or moving up a directory level like you did.

Noam-Dori, 2021-07-27:
I managed to replicate the same issue the other way, that is, if I loaded `/media/data/subt/virtual/install` first, it wouldnt load the packages, but when I changed to `/media/data/subt/virtual/install/share` everything loaded properly. The issue was that I reloaded the package index before the file index could update itself.

With the commit above, the index is loaded using an invalidation model, where external methods can only tell the index that it is incorrect. This way, the index can make its index when the file index it is based on is actually available.

Hopefully this commit solves the issue.
Cheers!

closed 2021-07-27 by Noam-Dori