Noam-Dori/ros-integrate #61: Launch file support

peci1, 2021-03-03:
Do you plan to support autocompletion and package resolving in launch files?

One can use 

```
<launch xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
        xsi:noNamespaceSchemaLocation="https://gist.githubusercontent.com/nalt/dfa2abc9d2e3ae4feb82ca5608090387/raw/roslaunch.xsd">
```

but that is a half-baked solution...

Noam-Dori, 2021-03-05:
Being such a fundemental component of ROS development, it is only a matter of time until this plugin will also support launch files

To be more specific, I plan to start working on it once I have the ROS infrastructure (which is only missing CMake and toolchain support) complete

While there is no real feature list yet, the plugin will offer support similar to package.xml files and run/debug capabilities

peci1, 2021-03-05:
Thanks! Looking forward for it :)

closed 2021-03-05 by Noam-Dori