Noam-Dori/ros-integrate #66: ArrayIndexOutOfBoundsException on each moce ov text cursor

vadimreutskiy, 2021-05-12:
**Describe the bug**
*A clear and concise description of what the bug is.*
I have quite a complex project which includes many ROS packages inside it. After enabling "ROS Support" plugin in the PyCharm IDE, it gets ArrayIndexOutOfBoundsException on each move of the text cursor.

The whole text of the error with call stack: 
java.lang.ArrayIndexOutOfBoundsException: Index 0 out of bounds for length 0
	at ros.integrate.pkg.ROSCompiledPackageFinder.getROSRoot(ROSCompiledPackageFinder.java:49)
	at ros.integrate.pkg.ROSCompiledPackageFinder.inFinder(ROSCompiledPackageFinder.java:134)
	at ros.integrate.pkg.ROSPackageFinderBase.lambda$investigate$1(ROSPackageFinderBase.java:98)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:176)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:497)
	at ros.integrate.pkg.ROSPackageFinderBase.investigate(ROSPackageFinderBase.java:101)
	at ros.integrate.pkg.ROSPackageManagerImpl.lambda$applyChangesToOrphans$3(ROSPackageManagerImpl.java:96)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:218)
	at com.intellij.openapi.application.TransactionGuardImpl.access$200(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$2.run(TransactionGuardImpl.java:200)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:781)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$invokeLater$4(ApplicationImpl.java:319)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:84)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:133)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:46)
	at com.intellij.openapi.application.impl.FlushQueue$FlushNow.run(FlushQueue.java:189)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:969)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:839)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:449)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:808)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$9(IdeEventQueue.java:448)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:781)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:496)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

**Expected behaviour**
No errors during the usage

**Environment Information:**
 - OS: Ubuntu 20.04.2 LTS 
 - IDE: PyCharm 2021.1, Professional version
 - Plugin Version: 0.1.6
 - Related Plugins: [hatchery](https://github.com/duckietown/hatchery)

Noam-Dori, 2021-05-12:
Hi @vadimreutskiy ,

This error is a bug caused by not having a ROS path (in the settings). It should not impact anything.
You can remove this error for now by filling in the path to your ROS distribution (in your case `/opt/ros/noetic`) in the ROS settings.
This error will not pop up in future versions at all as it has been fixed (issue #49 dealt with it) 

Also see issue #58 which reported the same bug

I hope this helps deal with the error popup

vadimreutskiy, 2021-05-24:
Thanks for the fast reply, Noam!

That solves the problem! 👍🏼

It will be very nice to have some notification with that advice directly in IDE, instead of the just array index error. However, I understand, that it is maybe not the first priority (and the opensource community can handle it by themselves) 😌

Noam-Dori, 2021-05-24:
> It will be very nice to have some notification with that advice directly in IDE, instead of the just array index error.

The issue with notifying people they have an empty ROS path is that often it is intentional, for instance, you are working on a non-ROS project on a non-ROS computer. The idea is that in the future, based on Windows/Ubuntu ROS infrastructures, the system will detect the `ROS Path` automatically and you won't need to care about filling it up

Anyways, I am happy the error was resolved
Cheers!

closed 2021-05-24 by Noam-Dori