Noam-Dori/ros-integrate #74: NullPointerException during index update

jessicaaustin, 2022-02-28:
**Describe the bug**

Plugin reports NPE and does not work at all. This started happening after I upgraded from pycharm 2020.3.3 to 2021.3.2

**To Reproduce**
Steps to reproduce the behavior:

1. Open project that has ROS files (like a msg file for example) in pycharm 2021.3.2
4. It doesn't seem like the plugin is working at all, and pycharm reports this error (see Stacktrace below)

Note I also tried invalidating all caches and restarting, still see this error.

**Environment Information:**
 - OS: Ubuntu 18.04
 - IDE: Pycharm 2021.3.2
 - Plugin Version: 0.1.6
 - Related Plugins: no other ROS plugins

**Stack Trace**

There are multiple errors reported, but they all have the same source.

```
java.lang.NullPointerException
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:35)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeMethod(MessageBusImpl.java:645)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeListener(MessageBusImpl.java:620)
	at com.intellij.util.messages.impl.MessageBusImpl.deliverMessage(MessageBusImpl.java:417)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpWaitingBuses(MessageBusImpl.java:390)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpMessages(MessageBusImpl.java:372)
	at com.intellij.util.messages.impl.MessageBusImpl.access$200(MessageBusImpl.java:33)
	at com.intellij.util.messages.impl.MessageBusImpl$MessagePublisher.invoke(MessageBusImpl.java:172)
	at com.sun.proxy.$Proxy21.fileWithNoDocumentChanged(Unknown Source)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at com.intellij.openapi.fileEditor.impl.FileDocumentManagerImpl.multiCast(FileDocumentManagerImpl.java:173)
	at com.intellij.openapi.fileEditor.impl.FileDocumentManagerImpl.lambda$new$0(FileDocumentManagerImpl.java:128)
	at com.sun.proxy.$Proxy21.fileWithNoDocumentChanged(Unknown Source)
	at com.intellij.openapi.fileEditor.impl.FileDocumentManagerImpl.contentsChanged(FileDocumentManagerImpl.java:681)
	at com.intellij.openapi.fileEditor.impl.FileDocumentManagerImpl$MyAsyncFileListener$1.afterVfsChange(FileDocumentManagerImpl.java:631)
	at com.intellij.openapi.vfs.newvfs.AsyncEventSupport.afterVfsChange(AsyncEventSupport.java:133)
	at com.intellij.openapi.vfs.newvfs.AsyncEventSupport$1.after(AsyncEventSupport.java:65)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeMethod(MessageBusImpl.java:645)
	at com.intellij.util.messages.impl.MessageBusImpl.invokeListener(MessageBusImpl.java:620)
	at com.intellij.util.messages.impl.MessageBusImpl.deliverMessage(MessageBusImpl.java:417)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpWaitingBuses(MessageBusImpl.java:390)
	at com.intellij.util.messages.impl.MessageBusImpl.pumpMessages(MessageBusImpl.java:372)
	at com.intellij.util.messages.impl.MessageBusImpl.access$200(MessageBusImpl.java:33)
	at com.intellij.util.messages.impl.MessageBusImpl$MessagePublisher.invoke(MessageBusImpl.java:179)
	at com.sun.proxy.$Proxy159.after(Unknown Source)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.fireAfterEvents(PersistentFSImpl.java:1188)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.applyMultipleEvents(PersistentFSImpl.java:1167)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.processEventsImpl(PersistentFSImpl.java:1130)
	at com.intellij.openapi.vfs.newvfs.AsyncEventSupport.processEventsFromRefresh(AsyncEventSupport.java:150)
	at com.intellij.openapi.vfs.newvfs.RefreshSessionImpl.fireEventsInWriteAction(RefreshSessionImpl.java:226)
	at com.intellij.openapi.vfs.newvfs.RefreshSessionImpl.lambda$fireEvents$1(RefreshSessionImpl.java:205)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$runEdtProgressWriteAction$11(ApplicationImpl.java:911)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$runProcess$2(CoreProgressManager.java:188)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$12(CoreProgressManager.java:624)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:698)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:646)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:623)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:66)
	at com.intellij.openapi.progress.impl.CoreProgressManager.runProcess(CoreProgressManager.java:175)
	at com.intellij.openapi.progress.util.PotemkinProgress.runInSwingThread(PotemkinProgress.java:164)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$runEdtProgressWriteAction$12(ApplicationImpl.java:911)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteActionWithClass(ApplicationImpl.java:935)
	at com.intellij.openapi.application.impl.ApplicationImpl.runEdtProgressWriteAction(ApplicationImpl.java:909)
[... 221 lines cut ...]
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:82)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:131)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:47)
	at com.intellij.openapi.application.impl.FlushQueue$FlushNow.run(FlushQueue.java:187)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:891)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:760)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:447)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:818)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:446)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:805)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:492)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

jessicaaustin, 2022-02-28:
After looking at the repo, it seems versions after 2020 might not be supported. In which case maybe the "bug" is that it's not clear which versions are supported by this plugin.

Noam-Dori, 2022-02-28:
This stack trace seems to be the same encountered issue #68 . From what I said when that issue was fixed:

> ... all plugin services (save for few plugin components) were not loaded by the plugin.xml.
> This was caused by an improper use of XInclude in the plugin.xml. XInclude, which allows importing tags from other documents, should be used at the root on the including document, and not deeper within the tree (which I did). Fixing this problem allowed the components to load.
>
> The fix should be available for the next release, 0.1.7, and is already available here.
>
> Fix was tested on IntelliJ IDEA Community 2021.2 on Windows 10.

If you are up for it, you can [install the plugin from source](https://github.com/Noam-Dori/ros-integrate/wiki/Working-From-Source#guide---install-ros-support-from-source) to apply the fix right now, which will fix the bug.

jessicaaustin, 2022-02-28:
Ah, I didn't realize that was the same issue. I'll try building myself and report back. Thanks!