Noam-Dori/ros-integrate #77: Broken in CLion 2022.3.2

stijnb1234, 2023-02-07:
**Describe the bug**
In the latest release version of CLion, the plugin does not even load. Can you check that out?

Noam-Dori, 2023-02-07:
Hi stijnb1234,
do you get any error message when it does not load? A stack trace can be very useful in tacking the issue

stijnb1234, 2023-02-07:
The first error I get:
```
Unable to save plugin settings
The plugin ros-integrate failed to save settings and has been disabled. Please restart CLION.
```

Then I get this stacktrace:
```
CompositeException (2 nested):
------------------------------
[1]: java.lang.RuntimeException: Cannot invoke (class=ROSIndexChangeListener, method=after, topic=BulkFileListener)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:639)
	at com.intellij.util.messages.impl.MessageBusImplKt.executeOrAddToQueue(MessageBusImpl.kt:466)
	at com.intellij.util.messages.impl.ToDirectChildrenMessagePublisher.publish$intellij_platform_core(CompositeMessageBus.kt:295)
	at com.intellij.util.messages.impl.MessagePublisher.invoke(MessageBusImpl.kt:421)
	at jdk.proxy2/jdk.proxy2.$Proxy126.after(Unknown Source)
	at com.intellij.util.FileContentUtilCore.lambda$reparseFiles$0(FileContentUtilCore.java:47)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:980)
	at com.intellij.util.FileContentUtilCore.reparseFiles(FileContentUtilCore.java:35)
	at com.jetbrains.cidr.lang.symbols.symtable.FileSymbolTablesCache.updateDirtyFilesPSI(FileSymbolTablesCache.java:538)
	at com.jetbrains.cidr.lang.symbols.symtable.FileSymbolTablesCache.reparseCachedPsiFiles(FileSymbolTablesCache.java:502)
	at com.jetbrains.cidr.lang.symbols.symtable.building.OCClearingSymbolsActivity.lambda$build$0(OCClearingSymbolsActivity.java:32)
	at com.jetbrains.cidr.lang.symbols.symtable.building.OCSymbolTableBuildingUtil.lambda$invokeAndWaitSafely$3(OCSymbolTableBuildingUtil.java:96)
	at com.intellij.openapi.application.WriteAction.lambda$run$1(WriteAction.java:86)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteActionWithClass(ApplicationImpl.java:968)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:994)
	at com.intellij.openapi.application.WriteAction.run(WriteAction.java:85)
	at com.jetbrains.cidr.lang.symbols.symtable.building.OCSymbolTableBuildingUtil.lambda$invokeAndWaitSafely$4(OCSymbolTableBuildingUtil.java:94)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:838)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:454)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:74)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:114)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:36)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:909)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:756)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$5(IdeEventQueue.java:437)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:787)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:436)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:615)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:434)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:838)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:480)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
Caused by: java.lang.NullPointerException: Cannot invoke "ros.integrate.pkg.ROSPackageManager.filesChanged(java.util.List)" because the return value of "com.intellij.openapi.project.Project.getService(java.lang.Class)" is null
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:35)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:655)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:621)
	... 47 more

[2]: java.lang.RuntimeException: Cannot invoke (class=ROSIndexChangeListener, method=after, topic=BulkFileListener)
[... cut ...]
Caused by: java.lang.NullPointerException: Cannot invoke "ros.integrate.pkg.ROSPackageManager.filesChanged(java.util.List)" because the return value of "com.intellij.openapi.project.Project.getService(java.lang.Class)" is null
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:35)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:655)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:621)
	... 47 more

------------------------------
```

Noam-Dori, 2023-02-07:
This stack trace seems to be similar to the one encountered in issue #68 . From what I said when that issue was fixed:

> ... all plugin services (save for few plugin components) were not loaded by the plugin.xml.
> This was caused by an improper use of XInclude in the plugin.xml. XInclude, which allows importing tags from other documents, should be used at the root on the including document, and not deeper within the tree (which I did). Fixing this problem allowed the components to load.
>
> The fix should be available for the next release, 0.1.7, and is already available here.
>
> Fix was tested on IntelliJ IDEA Community 2021.2 on Windows 10.

Did you [install the plugin from source](https://github.com/Noam-Dori/ros-integrate/wiki/Working-From-Source#guide---install-ros-support-from-source)? If not, you can do so, and it might fix the bug.
Let me know if the plugin still crashes

stijnb1234, 2023-02-07:
> This stack trace seems to be similar to the one encountered in issue #68 . From what I said when that issue was fixed:
>
>
> Did you [install the plugin from [...]

That resolved the problem. The version installed by CLion was 1.0.6 and the one from source was 1.0.7.

stijnb1234, 2023-02-07:
Received another stacktrace while saving a CMakeLists.txt file:

```
com.intellij.openapi.util.TraceableDisposable$DisposalException:         Library LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@6dc82cfb) already disposed:
        Library id: LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@6dc82cfb)
        Entity: workspace, null
        Is disposed in project model: true
        Stack trace: --------------Creation trace: 
java.lang.Throwable
	at com.intellij.openapi.util.TraceableDisposable.<init>(TraceableDisposable.java:27)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.<init>(LibraryBridgeImpl.kt:49)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:77)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:49)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:46)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl.createLibrary(ProjectLibraryTableBridgeImpl.kt:175)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$null$0(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$1(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1708)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:68)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.setDependency(ROSWorkspacePackageFinder.java:144)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$null$0(ROSIndexChangeListener.java:34)
	at com.intellij.util.containers.ContainerUtil$ImmutableListBackedByArray.forEach(ContainerUtil.java:495)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$after$1(ROSIndexChangeListener.java:34)
	at java.base/java.util.Spliterators$ArraySpliterator.forEachRemaining(Spliterators.java:992)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:762)
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:34)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:655)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:621)
	at com.intellij.util.messages.impl.MessageBusImplKt.executeOrAddToQueue(MessageBusImpl.kt:466)
	at com.intellij.util.messages.impl.ToDirectChildrenMessagePublisher.publish$intellij_platform_core(CompositeMessageBus.kt:295)
	at com.intellij.util.messages.impl.MessagePublisher.invoke(MessageBusImpl.kt:421)
	at jdk.proxy2/jdk.proxy2.$Proxy96.after(Unknown Source)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.fireAfterEvents(PersistentFSImpl.java:1208)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.processEvent(PersistentFSImpl.java:846)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.createChildFile(PersistentFSImpl.java:522)
	at com.intellij.openapi.vfs.newvfs.impl.VirtualFileSystemEntry.createChildData(VirtualFileSystemEntry.java:292)
	at com.intellij.configurationStore.StorageUtilKt.getOrCreateVirtualFile(storageUtil.kt:146)
	at com.intellij.configurationStore.FileBasedStorageKt.writeFile(FileBasedStorage.kt:280)
	at com.intellij.configurationStore.FileBasedStorage$FileSaveSession.saveLocally(FileBasedStorage.kt:117)
	at com.intellij.configurationStore.XmlElementStorage$XmlElementStorageSaveSession$SaveExecutor.save(XmlElementStorage.kt:165)
	at com.intellij.configurationStore.SaveExecutorKt.executeSave(SaveExecutor.kt:91)
	at com.intellij.configurationStore.SaveExecutorKt.saveSessions(SaveExecutor.kt:85)
	at com.intellij.configurationStore.ProjectSaveSessionProducerManager$saveWithAdditionalSaveSessions$saveResult$1.invokeSuspend$lambda$0(ProjectSaveSessionProducerManager.kt:31)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$runWriteAction$10(ApplicationImpl.java:989)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteActionWithClass(ApplicationImpl.java:968)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:989)
	at com.intellij.configurationStore.ProjectSaveSessionProducerManager$saveWithAdditionalSaveSessions$saveResult$1.invokeSuspend(ProjectSaveSessionProducerManager.kt:28)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at com.intellij.openapi.application.impl.DispatchedRunnable.run(DispatchedRunnable.kt:35)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:838)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
```

lithiumhydride2, 2023-02-24:
I install this plugin from source code. But is still crash under Clion 2022.3
```
com.intellij.openapi.util.TraceableDisposable$DisposalException:         Library LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@2b99be19) already disposed:
        Library id: LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@2b99be19)
        Entity: workspace, null
        Is disposed in project model: true
        Stack trace: --------------Creation trace: 
java.lang.Throwable
	at com.intellij.openapi.util.TraceableDisposable.<init>(TraceableDisposable.java:27)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.<init>(LibraryBridgeImpl.kt:49)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:77)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:49)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:46)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl.createLibrary(ProjectLibraryTableBridgeImpl.kt:175)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$0(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$1(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1708)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:68)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.setDependency(ROSWorkspacePackageFinder.java:144)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$after$0(ROSIndexChangeListener.java:34)
	at com.intellij.util.containers.ContainerUtil$ImmutableListBackedByArray.forEach(ContainerUtil.java:495)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$after$1(ROSIndexChangeListener.java:34)
	at java.base/java.util.Spliterators$ArraySpliterator.forEachRemaining(Spliterators.java:992)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:762)
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:34)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:655)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:621)
	at com.intellij.util.messages.impl.MessageBusImplKt.executeOrAddToQueue(MessageBusImpl.kt:466)
	at com.intellij.util.messages.impl.ToDirectChildrenMessagePublisher.publish$intellij_platform_core(CompositeMessageBus.kt:295)
	at com.intellij.util.messages.impl.MessagePublisher.invoke(MessageBusImpl.kt:421)
	at jdk.proxy2/jdk.proxy2.$Proxy172.after(Unknown Source)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.fireAfterEvents(PersistentFSImpl.java:1208)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl$3.close(PersistentFSImpl.java:801)
	at java.base/sun.nio.cs.StreamEncoder.implClose(StreamEncoder.java:347)
	at java.base/sun.nio.cs.StreamEncoder.close(StreamEncoder.java:169)
	at java.base/java.io.OutputStreamWriter.close(OutputStreamWriter.java:252)
	at java.base/java.io.BufferedWriter.close(BufferedWriter.java:269)
	at kotlin.io.CloseableKt.closeFinally(Closeable.kt:56)
	at com.intellij.configurationStore.StringDataWriter.write(XmlElementStorage.kt:430)
	at com.intellij.configurationStore.DataWriter.write$default(XmlElementStorage.kt:411)
	at com.intellij.configurationStore.FileBasedStorageKt.doWrite(FileBasedStorage.kt:351)
	at com.intellij.configurationStore.FileBasedStorageKt.writeFile(FileBasedStorage.kt:303)
	at com.intellij.configurationStore.FileBasedStorage$FileSaveSession.saveLocally(FileBasedStorage.kt:117)
	at com.intellij.configurationStore.XmlElementStorage$XmlElementStorageSaveSession$SaveExecutor.save(XmlElementStorage.kt:165)
	at com.intellij.configurationStore.SaveExecutorKt.executeSave(SaveExecutor.kt:91)
	at com.intellij.configurationStore.SaveExecutorKt.saveSessions(SaveExecutor.kt:85)
	at com.intellij.configurationStore.ProjectSaveSessionProducerManager$saveWithAdditionalSaveSessions$saveResult$1.invokeSuspend$lambda$0(ProjectSaveSessionProducerManager.kt:31)
	at com.intellij.openapi.application.impl.ApplicationImpl.lambda$runWriteAction$10(ApplicationImpl.java:989)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteActionWithClass(ApplicationImpl.java:968)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:989)
	at com.intellij.configurationStore.ProjectSaveSessionProducerManager$saveWithAdditionalSaveSessions$saveResult$1.invokeSuspend(ProjectSaveSessionProducerManager.kt:28)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at com.intellij.openapi.application.impl.DispatchedRunnable.run(DispatchedRunnable.kt:35)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
[... cut ...]
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:702)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:699)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1.run(Executors.java:699)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

stijnb1234, 2023-02-24:
Jup same issue

lithiumhydride2, 2023-02-24:
Sorry, I deleted it.

stijnb1234, 2023-02-24:
> Sorry, I deleted it.

Why?

stijnb1234, 2023-03-06:
@Noam-Dori Could you check this out?

Noam-Dori, 2023-03-15:
Hi @stijnb1234 and @lithiumhydride2 ,

Sorry it took me so long to respond

From my experience, the "already disposed" simply means the plugin tries to delete a library that was already deleted, and should not harm the plugin or anything else.

In any case, I tried running it with CLion 2022.3.2 IntelliJ Community 2022.3.2 but did not recieve the already disposed exception. To ensure compatibility, I also made the above commit which allows compilation to properly work for those versions. Does the commit fix those issues?

stijnb1234, 2023-03-15:
Thanks! I've updated the plugin in my CLion IDE and the exception seems to be gone.

Note, please update [the Build from source wiki page](https://github.com/Noam-Dori/ros-integrate/wiki/Working-From-Source#guide---install-ros-support-from-source to mention) to mention the requirement of JDK 17.

stijnb1234, 2023-03-17:
Got another error when opening a not-ROS project.

```
com.intellij.serviceContainer.AlreadyDisposedException: Cannot create com.intellij.workspaceModel.ide.WorkspaceModel because container is already disposed (container=Project(name=build, containerState=DISPOSE_COMPLETED, componentStore=/home/stijn/Documenten/School/WOR-World/ROSRobotArm/build) (disposed))
	at com.intellij.serviceContainer.ContainerUtilKt.throwAlreadyDisposedError(containerUtil.kt:40)
	at com.intellij.serviceContainer.ComponentManagerImpl.doGetService(ComponentManagerImpl.kt:695)
	at com.intellij.serviceContainer.ComponentManagerImpl.getService(ComponentManagerImpl.kt:629)
	at com.intellij.workspaceModel.ide.WorkspaceModel$Companion.getInstance(WorkspaceModel.kt:99)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.checkDisposed(LibraryBridgeImpl.kt:155)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.getLibrarySnapshot(LibraryBridgeImpl.kt:80)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.getFiles(LibraryBridgeImpl.kt:109)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getWorkspaceRoots(ROSWorkspacePackageFinder.java:61)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.inFinder(ROSWorkspacePackageFinder.java:173)
	at ros.integrate.pkg.ROSPackageFinderBase.lambda$investigate$1(ROSPackageFinderBase.java:98)
```

Noam-Dori, 2023-05-01:
Hi @stijnb1234,

I made the above commit to fix the error you posted above. Does it fix it?

Noam-Dori, 2023-05-01:
Closing this issue as the bug was resolved in the above commit (see #81)

closed 2023-05-01 by Noam-Dori