Noam-Dori/ros-integrate #79: More usage info?

stijnb1234, 2023-03-15:
I'm a bit unsure how to use the plugin.

Which steps should I follow to create a ROS project?
Is CLion recommended, or should I use IntelliJ IDEA instead?
How to use the creation dialogs (messages, ...)?

Noam-Dori, 2023-03-15:
The idea behind the plugin is that it doesn't force you to work differently than you already do now, and instead gives good support.
The plugin is still in its infancy, but here are things the plugin can help with:
- The plugin is able to detect ROS packages in your workspace and external libraries (for example, ros_noetic), and marks them accordingly.
- using the detected packages as an index, working with message files is significantly easier: the plugin adds custom highlighting and autocompletion, as well as linking between the various message types like CLion links classes. Full details can be found in the corresponding [wiki page](https://github.com/Noam-Dori/ros-integrate/wiki/Interfaces).
- The plugin also adds full support for package.xml files. It has tools that help complete the package.xml file, provide autocompletion for dependencies, as well as detect issues with it, and fix them just like CLion offers to fix common C++ issues. One of the interesting features is that if you autocomplete a message type linking to another package (for example, in `my_pkg/FootedQuaternion`, if I link to `geometry_msgs/Quaternion`, then `my_pkg/package.xml` will automatically add `geometry_msgs` as a dependency.
There are more features, but the idea is that it works behind the scenes so that you have to do less work on your projects.

Now to your questions:
1. The plugin works with all IntelliJ based platforms, so the platform depends on the language you want to use within ROS. Usually that's C++/CMake, so CLion is the best IDE for that (CLion users are also the majority of people who download this plugin). If you want to code solely in python, then PyCharm works best. If you want to code in Java using rosjava, then use IntelliJ.
2. If you are asking in what contexts to call the message creation dialogs, at least untill CMake support is added, the dialogue is only neccesary if you are creating a new message from the quick-fix (again, see the [interface wiki page](https://github.com/Noam-Dori/ros-integrate/wiki/Interfaces))
3. If you just want to run a node, CLion and PyCharm already do that, and you can [look at their guides](https://github.com/Noam-Dori/ros-integrate/wiki) as reference. If you want to run launch files, the plugin does not currently support that, but it hopefully will in the future (see issue #67)

stijnb1234, 2023-03-15:
I see, thanks for your quick reply! The ROS CLion tutorial you linked to (the first one) recommends installing the ROS-Robot Operating System plugin. Is this recommended by you too / is it compatible with this plugin? I guess it makes creating a ROS project easier?

Noam-Dori, 2023-03-15:
To my knowledge it is compatible since they do different things, but I cannot say for certain. Whether you should or should not use it (or any plugin really) is up to you.

stijnb1234, 2023-03-15:
> To my knowledge it is compatible since they do different things, but I cannot say for certain. Whether you should or should not use it (or any plugin really) is up to you.

I see, thanks for your info. If any more questions pop up, I will let you know.

closed 2023-03-15 by stijnb1234