Noam-Dori/ros-integrate #81: Broken in CLion 2023.1

peci1, 2023-04-11:
**Describe the bug**
I decided to give the plugin another go seeing the current development efforts. But it does nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the plugin to CLion, restart
2. Open a project
3. Configure ROS workspace in settings, restart
4. The plugin does nothing except syntax highlighting in .msg files. Package.xml files have no added deatures, no package name suggestions, licenses etc.

**Environment Information:**
 - OS: Ubuntu 18.04
 - IDE: CLion 2023.1
 - Plugin Version: master branch
 - Related Plugins: none

**Stack Trace**

This is a huge one:

<details>

```
com.intellij.openapi.util.TraceableDisposable$DisposalException:         Library LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@205f311f) already disposed:
        Library id: LibraryId(name=workspace, tableId=com.intellij.workspaceModel.storage.bridgeEntities.LibraryTableId$ProjectLibraryTableId@205f311f)
        Entity: workspace, null
        Is disposed in project model: true
        Stack trace: --------------Creation trace: 
java.lang.Throwable
	at com.intellij.openapi.util.TraceableDisposable.<init>(TraceableDisposable.java:27)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.<init>(LibraryBridgeImpl.kt:54)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:73)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:45)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.createLibrary(ProjectModifiableLibraryTableBridgeImpl.kt:42)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl.createLibrary(ProjectLibraryTableBridgeImpl.kt:176)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$0(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$1(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1708)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:68)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.setDependency(ROSWorkspacePackageFinder.java:144)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$after$0(ROSIndexChangeListener.java:34)
	at com.intellij.util.containers.ContainerUtil$ImmutableListBackedByArray.forEach(ContainerUtil.java:617)
	at ros.integrate.pkg.ROSIndexChangeListener.lambda$after$1(ROSIndexChangeListener.java:34)
	at java.base/java.util.Spliterators$ArraySpliterator.forEachRemaining(Spliterators.java:992)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:762)
	at ros.integrate.pkg.ROSIndexChangeListener.after(ROSIndexChangeListener.java:34)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:680)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:644)
	at com.intellij.util.messages.impl.MessageBusImplKt.executeOrAddToQueue(MessageBusImpl.kt:482)
	at com.intellij.util.messages.impl.ToDirectChildrenMessagePublisher.publish$intellij_platform_core(CompositeMessageBus.kt:294)
	at com.intellij.util.messages.impl.MessagePublisher.invoke(MessageBusImpl.kt:442)
	at jdk.proxy2/jdk.proxy2.$Proxy93.after(Unknown Source)
	at com.intellij.util.FileContentUtilCore.lambda$reparseFiles$0(FileContentUtilCore.java:47)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:958)
	at com.intellij.util.FileContentUtilCore.reparseFiles(FileContentUtilCore.java:35)
	at com.intellij.openapi.file.exclude.PersistentFileSetManager.lambda$onFileSettingsChanged$1(PersistentFileSetManager.java:82)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:831)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:456)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:79)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:122)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:41)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:789)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:740)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:734)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
[... 371 lines cut ...]
	at com.intellij.openapi.progress.util.BackgroundTaskUtil.runUnderDisposeAwareIndicator(BackgroundTaskUtil.java:361)
	at com.intellij.openapi.actionSystem.impl.ActionUpdater.lambda$doExpandActionGroupAsync$17(ActionUpdater.java:391)
	at io.opentelemetry.context.Context.lambda$wrap$1(Context.java:212)
	at com.intellij.util.concurrency.BoundedTaskExecutor.doRun(BoundedTaskExecutor.java:248)
	at com.intellij.util.concurrency.BoundedTaskExecutor.access$200(BoundedTaskExecutor.java:28)
	at com.intellij.util.concurrency.BoundedTaskExecutor$1.executeFirstTaskAndHelpQueue(BoundedTaskExecutor.java:221)
	at com.intellij.util.ConcurrencyUtil.runUnderThreadName(ConcurrencyUtil.java:227)
	at com.intellij.util.concurrency.BoundedTaskExecutor$1.run(BoundedTaskExecutor.java:210)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:702)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:699)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1.run(Executors.java:699)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

</details>

**Screenshots**
I can see the "ROS" item appear shortly in External libraries on startup, but then it disappears and never appears again.

https://user-images.githubusercontent.com/182533/231294264-3c9af764-53a5-477a-804f-50e502f7f8e0.mp4

Another issue - when I right-click a .msg file and choose "Override file type", I get 3 options named "ROSPkt (from 'ROS Support' plugin)". This is not very helpful.

peci1, 2023-04-11:
The same happens in Pycharm Pro 2023.1.

peci1, 2023-04-11:
And another exception:

<details>

```
java.lang.Throwable: Assertion failed: Write access is allowed inside write-action only (see com.intellij.openapi.application.Application.runWriteAction())
	at com.intellij.openapi.diagnostic.Logger.assertTrue(Logger.java:258)
	at com.intellij.openapi.application.impl.ApplicationImpl.assertWriteAccessAllowed(ApplicationImpl.java:1284)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl.updateProjectModel(WorkspaceModelImpl.kt:107)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.commit(ProjectModifiableLibraryTableBridgeImpl.kt:100)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl.createLibrary(ProjectLibraryTableBridgeImpl.kt:177)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$0(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$1(ROSWorkspacePackageFinder.java:70)
	at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1740)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:68)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getWorkspaceRoots(ROSWorkspacePackageFinder.java:61)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getScope(ROSWorkspacePackageFinder.java:95)
	at ros.integrate.pkg.ROSPackageFinderBase.findAndCache(ROSPackageFinderBase.java:40)
	at ros.integrate.pkg.ROSPackageManagerImpl.lambda$findPackages$1(ROSPackageManagerImpl.java:63)
	at com.intellij.util.containers.ContainerUtil$ImmutableListBackedByArray.forEach(ContainerUtil.java:617)
	at ros.integrate.pkg.ROSPackageManagerImpl.findPackages(ROSPackageManagerImpl.java:63)
	at ros.integrate.pkg.ROSPackageManagerImpl.loadIndex(ROSPackageManagerImpl.java:50)
	at ros.integrate.pkg.ROSPackageManagerImpl.findPackage(ROSPackageManagerImpl.java:174)
	at ros.integrate.pkg.ROSPackageIconProvider.getIcon(ROSPackageIconProvider.java:22)
	at com.intellij.ide.projectView.impl.CompoundIconProvider.getIcon(CompoundIconProvider.java:29)
	at com.intellij.ide.projectView.impl.CompoundIconProvider.findIcon(CompoundIconProvider.java:52)
	at com.intellij.ide.projectView.impl.nodes.PsiDirectoryNode.setupIcon(PsiDirectoryNode.java:184)
	at com.intellij.ide.projectView.impl.nodes.PsiDirectoryNode.updateImpl(PsiDirectoryNode.java:141)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.lambda$doUpdate$1(AbstractPsiBasedNode.java:204)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:865)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.doUpdate(AbstractPsiBasedNode.java:157)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.lambda$update$0(AbstractPsiBasedNode.java:153)
	at com.intellij.util.AstLoadingFilter.lambda$toComputable$2(AstLoadingFilter.java:172)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:130)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:119)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:109)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.update(AbstractPsiBasedNode.java:153)
	at com.intellij.ide.util.treeView.PresentableNodeDescriptor.getUpdatedPresentation(PresentableNodeDescriptor.java:90)
	at com.intellij.ide.util.treeView.PresentableNodeDescriptor.update(PresentableNodeDescriptor.java:33)
	at com.intellij.ui.tree.StructureTreeModel$Node.update(StructureTreeModel.java:485)
	at com.intellij.ui.tree.StructureTreeModel.invalidateInternal(StructureTreeModel.java:243)
	at com.intellij.ui.tree.StructureTreeModel.lambda$invalidate$5(StructureTreeModel.java:207)
	at com.intellij.ui.tree.StructureTreeModel.lambda$onValidThread$2(StructureTreeModel.java:142)
	at com.intellij.ui.tree.StructureTreeModel.lambda$onValidThread$1(StructureTreeModel.java:120)
	at com.intellij.util.concurrency.Invoker$Task.run(Invoker.java:304)
	at com.intellij.openapi.application.impl.ApplicationImpl.tryRunReadAction(ApplicationImpl.java:1102)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.lambda$runInReadActionWithWriteActionPriority$0(ProgressIndicatorUtils.java:72)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtilService.runActionAndCancelBeforeWrite(ProgressIndicatorUtilService.java:63)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.runActionAndCancelBeforeWrite(ProgressIndicatorUtils.java:129)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.lambda$runWithWriteActionPriority$1(ProgressIndicatorUtils.java:110)
	at com.intellij.openapi.progress.ProgressManager.lambda$runProcess$1(ProgressManager.java:70)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$runProcess$2(CoreProgressManager.java:186)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$13(CoreProgressManager.java:604)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:679)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:635)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:603)
[... cut ...]
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:702)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:699)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1.run(Executors.java:699)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

</details>

Noam-Dori, 2023-04-30:
I replicated the error.

The error is unsurprisingly caused because the library is being disposed, then the disposed library is used again, which causes errors like a NPE would, hence why nothing is working.
The surprising thing is that the very creation of the library caused its disposal as it triggered a module root update event.

The code no longer stores libraries as they are constantly disposed, and the fix was tested on CLion 2023.1
I want to test it on PyCharm and IntelliJ before committing it though.

Noam-Dori, 2023-05-01:
I believe the major bug was dealt with by the above commit. Does it fix the issues you encountered?

peci1, 2023-05-01:
Playing with Additional package path seems to semi-randomly switch between the modes. Sometimes even Reload CMake project does. However, there is no pattern in it - even two projects from the same metapackage (and workspace): one has mode 1 when I clear the additional packages path, while the other has mode 1 when I fill the source path to the underlying workspace.

peci1, 2023-05-01:
When opening a project that was never opened with this plugin, I also get assertion error:

<details>

```
java.lang.Throwable: Assertion failed: Write access is allowed inside write-action only (see com.intellij.openapi.application.Application.runWriteAction())
	at com.intellij.openapi.diagnostic.Logger.assertTrue(Logger.java:258)
	at com.intellij.openapi.application.impl.ApplicationImpl.assertWriteAccessAllowed(ApplicationImpl.java:1284)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.LibraryBridgeImpl.setEntityStorage(LibraryBridgeImpl.kt:60)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl$1.changed(ProjectLibraryTableBridgeImpl.kt:70)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeMethod(MessageBusImpl.kt:680)
	at com.intellij.util.messages.impl.MessageBusImplKt.invokeListener(MessageBusImpl.kt:640)
	at com.intellij.util.messages.impl.MessageBusImplKt.executeOrAddToQueue(MessageBusImpl.kt:482)
	at com.intellij.util.messages.impl.MessagePublisher.publish$intellij_platform_core(MessageBusImpl.kt:465)
	at com.intellij.util.messages.impl.MessagePublisher.invoke(MessageBusImpl.kt:442)
	at jdk.proxy2/jdk.proxy2.$Proxy63.changed(Unknown Source)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl$onChanged$2.invoke(WorkspaceModelImpl.kt:281)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl$onChanged$2.invoke(WorkspaceModelImpl.kt:280)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl.logErrorOnEventHandling(WorkspaceModelImpl.kt:318)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl.onChanged(WorkspaceModelImpl.kt:280)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl.access$onChanged(WorkspaceModelImpl.kt:32)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl$updateProjectModel$generalTime$1$7.invoke(WorkspaceModelImpl.kt:141)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl$updateProjectModel$generalTime$1$7.invoke(WorkspaceModelImpl.kt:141)
	at com.intellij.workspaceModel.storage.impl.VersionedEntityStorageImpl.replace(VersionedEntityStorageImpl.kt:183)
	at com.intellij.workspaceModel.ide.impl.WorkspaceModelImpl.updateProjectModel(WorkspaceModelImpl.kt:141)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectModifiableLibraryTableBridgeImpl.commit(ProjectModifiableLibraryTableBridgeImpl.kt:100)
	at com.intellij.workspaceModel.ide.impl.legacyBridge.library.ProjectLibraryTableBridgeImpl.createLibrary(ProjectLibraryTableBridgeImpl.kt:177)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.lambda$getLibrary$0(ROSWorkspacePackageFinder.java:66)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:66)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getWorkspaceRoots(ROSWorkspacePackageFinder.java:58)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getScope(ROSWorkspacePackageFinder.java:90)
	at ros.integrate.pkg.ROSPackageFinderBase.findAndCache(ROSPackageFinderBase.java:40)
	at ros.integrate.pkg.ROSPackageManagerImpl.lambda$findPackages$1(ROSPackageManagerImpl.java:63)
	at com.intellij.util.containers.ContainerUtil$ImmutableListBackedByArray.forEach(ContainerUtil.java:617)
	at ros.integrate.pkg.ROSPackageManagerImpl.findPackages(ROSPackageManagerImpl.java:63)
	at ros.integrate.pkg.ROSPackageManagerImpl.loadIndex(ROSPackageManagerImpl.java:50)
	at ros.integrate.pkg.ROSPackageManagerImpl.findPackage(ROSPackageManagerImpl.java:174)
	at ros.integrate.pkg.ROSPackageIconProvider.getIcon(ROSPackageIconProvider.java:22)
	at com.intellij.ide.projectView.impl.CompoundIconProvider.getIcon(CompoundIconProvider.java:29)
	at com.intellij.ide.projectView.impl.CompoundIconProvider.findIcon(CompoundIconProvider.java:52)
	at com.intellij.ide.projectView.impl.nodes.PsiDirectoryNode.setupIcon(PsiDirectoryNode.java:184)
	at com.intellij.ide.projectView.impl.nodes.PsiDirectoryNode.updateImpl(PsiDirectoryNode.java:124)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.lambda$doUpdate$1(AbstractPsiBasedNode.java:204)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:865)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.doUpdate(AbstractPsiBasedNode.java:157)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.lambda$update$0(AbstractPsiBasedNode.java:153)
	at com.intellij.util.AstLoadingFilter.lambda$toComputable$2(AstLoadingFilter.java:172)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:130)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:119)
	at com.intellij.util.AstLoadingFilter.disallowTreeLoading(AstLoadingFilter.java:109)
	at com.intellij.ide.projectView.impl.nodes.AbstractPsiBasedNode.update(AbstractPsiBasedNode.java:153)
	at com.intellij.ide.util.treeView.PresentableNodeDescriptor.getUpdatedPresentation(PresentableNodeDescriptor.java:90)
	at com.intellij.ide.util.treeView.PresentableNodeDescriptor.update(PresentableNodeDescriptor.java:33)
	at com.intellij.ui.tree.StructureTreeModel$Node.update(StructureTreeModel.java:485)
	at com.intellij.ui.tree.StructureTreeModel$Node.<init>(StructureTreeModel.java:466)
	at com.intellij.ui.tree.StructureTreeModel$Node.<init>(StructureTreeModel.java:459)
	at com.intellij.ui.tree.StructureTreeModel.getValidChildren(StructureTreeModel.java:417)
[... cut ...]
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:702)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1$1.run(Executors.java:699)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.util.concurrent.Executors$PrivilegedThreadFactory$1.run(Executors.java:699)
	at java.base/java.lang.Thread.run(Thread.java:833)
```

</details>

peci1, 2023-05-01:
And I got this once (but only once):

<details>

```
com.intellij.serviceContainer.AlreadyDisposedException: Cannot create com.intellij.openapi.roots.impl.libraries.ProjectLibraryTable because container is already disposed (container=Project(name=roboton_description, containerState=DISPOSE_COMPLETED, componentStore=/media/data/roboton/ws/src/roboton-ros/roboton_description) (disposed))
	at com.intellij.serviceContainer.ContainerUtilKt.throwAlreadyDisposedError(containerUtil.kt:40)
	at com.intellij.serviceContainer.ComponentManagerImpl.doGetService(ComponentManagerImpl.kt:736)
	at com.intellij.serviceContainer.ComponentManagerImpl.getService(ComponentManagerImpl.kt:670)
	at com.intellij.openapi.roots.impl.libraries.LibraryTablesRegistrarImpl.getLibraryTable(LibraryTablesRegistrarImpl.java:45)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getLibrary(ROSWorkspacePackageFinder.java:65)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.getWorkspaceRoots(ROSWorkspacePackageFinder.java:58)
	at ros.integrate.pkg.ROSWorkspacePackageFinder.inFinder(ROSWorkspacePackageFinder.java:168)
	at ros.integrate.pkg.ROSPackageFinderBase.lambda$investigate$1(ROSPackageFinderBase.java:98)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:178)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1845)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:596)
	at ros.integrate.pkg.ROSPackageFinderBase.investigate(ROSPackageFinderBase.java:101)
	at ros.integrate.pkg.ROSPackageManagerImpl.lambda$applyChangesToOrphans$3(ROSPackageManagerImpl.java:96)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:831)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:456)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:79)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:122)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:41)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:788)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:739)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:731)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:758)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.kt:666)
	at com.intellij.ide.IdeEventQueue._dispatchEvent$lambda$7(IdeEventQueue.kt:570)
	at com.intellij.openapi.application.impl.ApplicationImpl.withoutImplicitRead(ApplicationImpl.java:1446)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.kt:570)
	at com.intellij.ide.IdeEventQueue.access$_dispatchEvent(IdeEventQueue.kt:68)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:349)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:348)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:787)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:348)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:343)
	at com.intellij.ide.IdeEventQueueKt.performActivity$lambda$1(IdeEventQueue.kt:994)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.ide.IdeEventQueueKt.performActivity(IdeEventQueue.kt:994)
	at com.intellij.ide.IdeEventQueue.dispatchEvent$lambda$4(IdeEventQueue.kt:343)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:831)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.kt:385)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
```

</details>

Noam-Dori, 2023-05-01:
I think I can help you with the pattern behind this flavor of issues.

Since packages are completely independent from the workspace scope (the workspace doesn't have to be the project in its entirety, and obviously the core ROS files are outside the workspace), the plugin needs to create different libraries which are in turn used to run searches and indexing.

When the "workspace" library (`.idea/libraries/workspace.xml`) fails (for example the previous error), the plugin is not applied on the project at all, hence "mode 3"

When the "ROS" library (`.idea/libraries/ROS.xml`) fails, all the ROS defined packages cannot be looked for and are treated as if they don't exist, hence "mode 2" (assuming the workspace is loaded properly)

Adding additional package paths extends the workspace library, so that would explain why it works. From my experience the rest of the code works fine, but the library indexing has many issues. The bugs you raise (and raised in the past) help me find those issues which is something I appreciate.

Message files are one of the few features that do not rely on packages at all, so that explains why they are the only things that work when the libraries don't work

> In mode 2, when I hover or Ctrl+Click a message definition in a message file, I get an NPE:

Was the message you clicked on a "ROS" message or from your workspace? If it was a ROS default message, that would explain why it would crash.

peci1, 2023-05-01:
> Was the message you clicked on a "ROS" message or from your workspace? If it was a ROS default message, that would explain why it would crash.

The message I tried had both standard library types (Header) and custom types. Header was not red and caused the NPE. The custom messages were red (missing), and did not cause the NPE. I hope this helps...

Noam-Dori, 2023-05-01:
The first header in message files, just like in ROS, receives special treatment. The same applies in the plugin, and the plugin will not annotate the first header and will redirect to std_msgs/Header automatically, even if it does not exist (It should always succeed though as it is a standard ROS message)

peci1, 2023-05-01:
> When the "ROS" library (`.idea/libraries/ROS.xml`) fails

Hmm, but I don't see any errors logged...

Noam-Dori, 2023-05-01:
There are multiple ways the library can decide to not load, even without errors. For example, in the video you sent in this thread it fails without error: it loads (you can see it in the bottom of the external libraries tab) then immediately disappear, probably because it is being incorrectly disposed. I do not know why this happens, but I hope that once I manage to consistently replicate a scenario where this happens I will resolve it

Noam-Dori, 2023-05-03:
Since the principle bug reported here was fixed and CLion doesn't crash all the time, I think I will close this particular issue
My guess is that the other crashes and the "broken" states of the plugin are caused by the plugin attempting to load the libraries before it is aware of the files. For example, the "write access" exception when the plugin is opned on a new project is caused by the icon provider trying to determine whether a folder is a package or not because it is aware of the contents of the folder.

It seems there may be more sources of this sort of issue, and I want to tackle them all together. If you want to, you can write that stack-trace in issue #84 where I will tackle this issue as a whole

closed 2023-05-03 by Noam-Dori