Noam-Dori/ros-integrate #84: Components are loaded too early

Noam-Dori, 2023-05-03:
**Describe the bug**
Components are loaded too early by the plugin, which causes said components to crash.

**To Reproduce**
Open a project in PyCharm. You can obtain similar errors in other IDEs consistently by opening a project previously unregistered by the plugin.

**Expected behavior**
*Describe what you expected to happen.*

**Environment Information:**
 - OS: Bug seen in Ubuntu 20.04 and Windows 10
 - IDE: any supported IDE in version 2023.1
 - Plugin Version: master
 - Related Plugins: None.

**Additional context**
This can be caused by several components that cause the crash. The solution should be a generic solution that will ensure all components are loaded after the project is fully loaded.

Noam-Dori, 2023-08-23:
While I don't completely understand it, apparently UI threads like looking up icons are now canceled from within the code, potentially because of in-project changes, which prevent the index from loading correctly. To take care of this, the package index and libraries now load in their own "post startup activity" which loads them in the background. In the meantime, any component that depends on the packages (like the icons) either uses default behavior or waits until the packages are indexed.

Tested on PyCharm 2023.1 and IntelliJ IDEA 2023.1, both windows 10.

peci1, 2023-08-25:
Thanks for this update. It made things better. However, I still get some unsuccessful plugin loads in CLion. What seems to help now is changing anything in Settings->..->ROS. After that, all indexing and other goodies from the package start to work. However, I need to repeat that after every restart of CLion. No errors are reported (except for NPEs when hovering over red custom message types in .msg files).

Noam-Dori, 2023-08-25:
First, its reassuring that finally some progress was made.
What version of CLion are you using (and on what OS)?

peci1, 2023-08-25:
I'm running on RC channel (currently no RC, so 2023.2.1, but the same was yesterday with 2023.2). I'm using Ubuntu 18.04.

peci1, 2023-08-25:
I think what stopped happening is "mode 3" as mentioned in https://github.com/Noam-Dori/ros-integrate/issues/81#issuecomment-1529930460 .

Noam-Dori, 2023-08-25:
Tested on CLion 2023.1 and everything seems to work except for some icon loading, but that's purely visual
So, either the problem is due to Linux or 2023.2

Could you point to what exactly does not load in the plugin so I can test it directly?

peci1, 2023-08-25:
And do you try to load multiple projects (in multiple windows)?

For example, what I see sometimes now is that most files in the project are marked as excluded. When I remove the workspace path from ROS settings, the marking gets back to normal.

My usual workflow is: have a workspace with multiple packages and open some of the packages as CLion projects, each in its own window. Some packages are directly in `src/package`, some are in stacks, so in `src/stack/package`. For some projects, I also define Python interpreter, for some I don't. I launch CLion via the GUI shortcut, so no ROS variables from .bashrc are globally set. In the plugin settings, I set just the ROS path to /opt/ros/melodic and workspace path (the ws directly extends the base ROS workspace).

To use the same codebase, try to open a few packages from https://github.com/ctu-vras/ros-utils/ .

Noam-Dori, 2023-08-26:
I managed to replicate the bug in CLion 2023.1 on Windows 10. The main trigger for it was opening a package as a project that causes this, meaning something is wrong with how the "workspace" library is loaded. This would also explain why refreshing the index fixes it.

Working on figuring out the exact cause and solving it

Noam-Dori, 2023-08-27:
After this commit, what I see is that if the libraries are in order everything is fine. For some reason though, CLion cannot properly create the library on startup (it does after a settings change). This is a particular issue with the workspace library.

@peci1 does the commit make any progress on your end?

peci1, 2023-08-28:
Hmm, still not entirely fixed. The last step to reproduce I forgot to mention is to have a few project windows open, and then File->Exit to close the IDE, and then rerun the IDE, so that it reopens those few projects at once.

This is what actually breaks the plugin. Reopening the projects after startup is finished now seems to work as a sure fix. But it gets tiresome with 5-10 open projects...

[image]

Here you can see it's actually the ROS library that is broken, as the workspace packages are correctly recognized. Note in the left part of the image there is the end of the ROS library with none of the packages with the Compiled ROS Package icon.